srsran/srsRAN_4G #1240: open5gs connection error: Failed to initiate NG connection

jqhu27, 2023-11-03:
## Issue Description ##
I attempted to set up a 5G Standalone (SA) End-to-End scenario using open5gs as the core. I meticulously followed the step-by-step instructions provided in the documentation found [here](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_E2E/source/index.html) to deploy the scenario. However, when I attempted to execute the command `sudo srsenb`, I encountered the following error:
[image]

Subsequently, I investigated the open5gs **_amf.log_** file and observed the following:
[image]

I noticed that if I use the original open5gs **_amf.yaml_**, the **AMF** service can be started and runs without issues. However, when I attempt to use the modified **_amf.yaml_** provided on the srsRAN website, the service fails to start.

I suspect that there may be an issue with the **_amf.yaml_** configuration file. Could you kindly assist me in resolving this matter? Your help would be greatly appreciated. Thank you in advance.

Igormartinssilva, 2023-11-22:
I have the same problem and I don't know how to solve it either.

whitecloudy, 2023-12-27:
The Open5gs version has been updated from 2.6.6 to 2.7.0 recently. I think they changed some rules about config files. I have faced same issue that you showed and I fixed it by carefully changing enb's config file(enb.conf, rr.conf) which is related with 5g core connection. Mostly I changed was IP addresses. 
Read this article and find your way. I can't tell how I did exactly cause I barely tracked how I did.
https://open5gs.org/open5gs/docs/guide/01-quickstart/