srsran/srsRAN_4G #1241: srsUE does not ping to Open5GS-EPC-GW

AgustinVizcarra, 2023-11-05:
Hi dear community, the reason for the following email is to mentio that the following topology was implemented in OpenStack as shown in the following figure:
 
[image]

Where there are 3 virtual machines: One that has Open5GS-EPC-Core, while the other two correspond to srs-ENB and srs-UE. Adittionally, there are two network segments(one with Access to the internet and the other referring to the communication of the mobile network components). At the time of making the attachment everything seems to be going well according to the mme and UPF(SGWU) logs as shown in the following figures:
-mme.log:
 
[image]

-UPF(SGWU).log:
 
[image]

I also put the screenshots that show that the srsUE was able to correctly make an attachment to the network:
Tun_srsue:
 
[image]

However, when performin a ping that shows that the srsUE has Access to Internet, I have the following behavior:
 
[image]

To solve this I try doing the following:
1.	I capture the packets in the interface from the mme and I checked that for control plane there seemed to be no problem.
2.	I capture the packets in the interface corresponding to user plane (the ogstun interface) but in that case I noticed that I didn’t receive any packets in this interface. 
 
[image]

3.	So I decide to check the traffic output from the srsue (checking the tun_srsue) and I get the following:
 
[image]

I noticed that the traffic goes to cb-in-f139 and not by the ens3(management) or ens4 (mobile network), so the traffic keept in the vm which was corresponding to the srs_UE (And don’t know why)
4.	I checked if the ip route could be the problema behind these and I saw the following output:
 
[image]

Also it’s important to comment that the net.ipv4.ip_forward was setting on 1 (enable) (In the Open5GS-EPC VM and also in the srsUE VM)
5.	After that I didn’t know what to do or what could be the cause of this error.
Additionaly, I tell you that this configuration is based on the configurations made by [s5uishida](https://github.com/s5uishida) corresponding to [Open5GS EPC & srsRAN 4G with ZeroMQ UE / RAN Sample Configuration](https://github.com/s5uishida/open5gs_epc_srsran_sample_config) in which the only modifications made is that now the Open5Gs core exists in a single virtual machine (CUPS is not applied). Below attached in this email, I give you my configurations files,captures and screenshots taken.
 
Ogstun (Open5GS) screenshot:
 
[image]

Configurations Files and Captures Link:  [https://drive.google.com/file/d/1u9aCcDC1fLJ8JQIgXD9NHppybdXKi7x7/view?usp=sharing](url)

I await your response. 
Greetings

dewadragoon, 2024-04-29:
Hi, I am having the same issue.

my ue cant even ping to epc ip - 10.45.0.1.

the logs are similar with the above.

I would appreciate if anyone could point to the correct direction.

OmnneaRnD, 2025-01-21:
Hello Community..
Same here.. similar setup for 
- srsRAN (srsENB), 
- Open5GS (single vm for all Open5GS nodes/services), 

The difference is that I am using a SIM card on Samsung A25 5g phone as a UE 

same outcome: UE cannot ping 10.45.0.1, unable to access Internet!! 

Has this been resolved? 
Thanks.

aaleessa, 2025-01-21:
Hello Community..
Same here.. similar setup for

srsRAN (srsENB),
Open5GS (single vm for all Open5GS nodes/services),
The difference is that I am using a SIM card on Samsung A25 5g phone as a UE

same outcome: UE cannot ping 10.45.0.1, unable to access Internet!!

Has this been resolved?
Thanks.