srsran/srsRAN_4G #1243: Issues with Running srsENB on USRP E310

alinstoica19, 2023-11-12:
[image]
## Issue Description ##
I am encountering a persistent issue while attempting to run srsenb with a USRP E310 device. The process starts normally, but I consistently face a timeout error during control operations and issues with the End of Burst (EOB) signal. The specific error message is: [ERROR] [UHD RF] RfnocError: OpTimeout: Control operation timed out waiting for ACK followed by [INFO] [UHD RF] Tx while waiting for EOB, timed out... 6.46538 >= 0. Starting new burst....

## Setup Details ##
Operating System: Ubuntu 22.04(WSL)
Hardware: USRP E310

srsRAN Version: Version 23.4.0

  UHD (USRP Hardware Driver) Version: 4.5.0.0-0-g471af98f
   MPM Version: 5.3
   FPGA Version: 6.1
   FPGA git hash: a2a04e3.clean
   Time sources:  internal, external, gpsdo
  Clock sources: internal
  Sensors: ref_locked, gps_locked, temp_fpga, temp_mb, gps_gpgga, gps_sky, gps_time, gps_tpv

RF Front-end: 
 RX Frontend: 0
  Name: E3xx
  Antennas: RX2, TX/RX
  Freq range: 70.000 to 6000.000 MHz
  Gain range PGA: 0.0 to 76.0 step 1.0 dB
 Bandwidth range: 20000000.0 to 40000000.0 step 0.0 Hz
 Connection Type: IQ
 Uses LO offset: No

## Expected Behavior ##
The expectation is for srsENB to run without encountering timeout errors or issues related to EOB, establishing a stable LTE eNodeB.

## Actual Behaviour ##
During the execution of srsENB, the process fails to maintain stability, encountering a timeout error and issues with EOB. The error [ERROR] [UHD RF] RfnocError: OpTimeout: Control operation timed out waiting for ACK is consistently observed, and there are problems related to the End of Burst (EOB) signal

## Steps to reproduce the problem ##
Run srsENB with the USRP E310 connected and configured with the following arguments: addr=192.168.10.40,master_clock_rate=16e6.
Observe the error message and issues related to EOB during the execution.

chonghua1, 2024-06-11:
Hi, I encountered the same issue, have you resolved it?

alinstoica19, 2024-06-12:
Unfortunately not, my conclusion is that the e310 cannot be integrated with
srsRAN because of the few mega samples.