srsran/srsRAN_4G #1247: highly intermittent attach with srsenb and srsue with bladerf 2.0 micro

nitinvjain, 2023-11-21:
## Issue Description ##
I am referring to :
Document reference: https://buildmedia.readthedocs.org/media/pdf/srslte-docs/latest/srslte-docs.pdf

I am using srsRAN_4G along with bladeRF 2.0 micro Xa4 in a cabled setup as per following connections:
gNodeB Tx1 (40 dB tx_gain --> UE Rx1 (Agc)gnodeB Rx1 (Agc) <-- UE Tx1 (40 dB tx_gain)No GPSDO used for bladeRF.
Basically, I am using srsenb and srsue on 2 different machine.srsenb and srsepc are running on same machine.
Main issue is that attach is too intermittemt and 1 out of 10 times only, it happens. That too sometimes, it works without any release uptp 20-25 mins and sometimes, it releases within 2-3 mins itself.

I tried to play with time_adv_nsamples in UE side by changing it in steps of 5 right from 27 (default) to 140 but no change in behaviour. I am sharing all configurations by attching zip file having conf files as well as .log files.Please point out to me any obvious issues.My understanding based on logs is that DL is probably working fine and UL is quite flaky.Please share any insights to move forward.

I have also referred issues as per following details:
https://github.com/srsran/srsRAN_4G/issues/688

https://github.com/srsran/srsRAN_4G/issues/1105 
## Setup Details ##
Accordingly, I have changed default change in srate in source code and in ue.conf (srate=11.52 e6)
Other Configs:
bladeRF related:bladeRF  2.0 micro XA4bladeRF> version
  bladeRF-cli version:        1.8.0-git-unknown  libbladeRF version:         2.4.1-git-unknown
  Firmware version:           2.4.0-git-a3d5c55f  FPGA version:               0.14.0 (configured by USB host)
Ubuntu versions: Distributor ID: UbuntuDescription: Ubuntu 22.04.3 LTSRelease: 22.04Codename: jammy
srsRAN_4G:
git logcommit fa56836b14dc6ad7ce0c3484a1944ebe2cdbe63b (HEAD -> master, tag: release_23_04_1, origin/master, origin/HEAD)Merge: 921f17484 737c5ae71Author: Codebot <[codebot at softwareradiosystems.com](https://lists.srsran.com/mailman/listinfo/srsran-users)>Date:   Mon Jun 19 15:03:18 2023 +0200
    Merge branch 'next' into agpl_next
UE side: Intel i9 13900H gen based machine
gNodeB side: AMD Ryzen 

I also installed SoapySDR and SoapyBladeRF so that native soapySDR based driver can be used.I had to make changes in ue.conf and enb.conf: 
device_name = soapydevice_args = driver=bladerf
https://nickvsnetworking.com/srslte-install-for-bladerf-limesdr-on-debian-ubuntu/#:~:text=bladeRF/hostedx40.rbf%20popd%20git%20clone%20https://github.com/pothosware/SoapyBladeRF.git%20pushd%20SoapyBladeRF%20mkdir%20build%20cd%20build%20cmake%20..
[Describe the issue in detail]

## Expected Behavior ##
[Consistent attach is expected behaviour with srsenb and srsue]

## Actual Behaviour ##
[Attach is very much intermittent]

## Steps to reproduce the problem ##
[Described as above]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue
[srsRAN_4G-20231121T120116Z-001.zip](https://github.com/srsran/srsRAN_4G/files/13427206/srsRAN_4G-20231121T120116Z-001.zip)
]

nitinvjain, 2023-11-27:
Please refer to https://github.com/srsran/srsRAN_4G/issues/1248

Any solution available?

apptuser, 2024-01-16:
This is a blocker for several folks. Any updates or inputs from SRS engineers on this?

nitinvjain, 2024-01-17:
I think I can share a workaround to bypass this issue. Will post the fix. I do have a fork branch for this. Will share it. 
Regards Nitin  e-mail: ***@***.*** Cell #: +91 - 9242280803

nitinvjain, 2024-01-18:
Closing the issue here as root cause probably is change of sampling rate. That seems to cause some issue in BladeRF 2.0 micro either due to some bug in FPGA program or some other firmware issue. That never was addressed by BladeRF follks. So closing the issue as I am bypassing the issue by keeping sampling rate unchanged. This is not fully desirable as this simply means that cell search is running at much higher sampling rate. It is just a work around.

closed 2024-01-18 by nitinvjain

nitinvjain, 2024-01-18:
Hi All, 
I have closed the issue after posting relevant suggestion on a possible work around.https://github.com/srsran/srsRAN_4G/issues/1247Request srsRAN_4G to let me know if they would like to include this work around after due process.

Regards Nitin Jain 

 
 

 srs

apptuser, 2024-01-21:
Hi @nitinvjain. Realy appreciate your taking the time and posting the workaround. I am implemented the changes as you exactly had described and in your forked branch. Unfortunately, I was only able to get the UE attached twice. It is still highly intermittent. I am still struggling to get it stable. Is your setup stable and attaches everytime?

nitinvjain, 2024-01-22:
Hi,Thanks. Well, it is very consistent for me. No issues at all. Need to check your setup. To start with, can you please share your config files and changes which you have done.
Regards and cheersNitin

apptuser, 2024-01-23:
@nitinvjain as I played around with this more after your confirmation of consistent behavior, I ended up turning the wifi and bluetooth radiios off on my two laptops and now I am not seeing any issues anymore and consistent connection just as yours. Just to confirm I basically cloned your forked branch with no other modifications. So all good now, great work. Much appreciated.

nitinvjain, 2024-01-23:
> @nitinvjain as I played around with this more after your confirmation of consistent behavior, I ended up turning the wifi and bluetooth radiios off on my two laptops and now I am not seeing any [...]

Hi @apptuser . I am glad to hear that you could make it work. 
WIsh you all the best!!
Cheers!!