srsran/srsRAN_4G #1260: Make test got stuck for test 121: pssch_test_tm2_p6

reallsj, 2023-12-17:
System: Ubuntu 22.04.3 LTS
The previous steps are perfect for me, but when process make test, the 121 test got stuck. The terminal is still running, can anyone help?
ubuntu@sjli:~/srsRAN2/build$ make test
Running tests...
Test project /home/ubuntu/srsRAN2/build
          Start    1: thread_pool_test
   1/1523 Test    #1: thread_pool_test ................................................   Passed    4.14 sec
          Start    2: thread_test
   2/1523 Test    #2: thread_test .....................................................   Passed    0.00 sec
          Start    3: band_helper_test
   3/1523 Test    #3: band_helper_test ................................................   Passed    0.00 sec
          Start    4: pdu_test
   4/1523 Test    #4: pdu_test ........................................................   Passed    0.00 sec
          Start    5: mac_pcap_test
   5/1523 Test    #5: mac_pcap_test ...................................................   Passed    0.01 sec
          Start    6: mac_pdu_nr_test
   6/1523 Test    #6: mac_pdu_nr_test .................................................   Passed    0.00 sec
          Start    7: chest_test_dl_cellid0
   7/1523 Test    #7: chest_test_dl_cellid0 ...........................................   Passed    0.01 sec
          Start    8: chest_test_dl_cellid1
   8/1523 Test    #8: chest_test_dl_cellid1 ...........................................   Passed    0.00 sec
          Start    9: chest_test_dl_cellid2
   9/1523 Test    #9: chest_test_dl_cellid2 ...........................................   Passed    0.00 sec
          Start   10: chest_test_dl_cellid0_50prb
  10/1523 Test   #10: chest_test_dl_cellid0_50prb .....................................   Passed    0.01 sec
          Start   11: chest_test_dl_cellid1_50prb
  11/1523 Test   #11: chest_test_dl_cellid1_50prb .....................................   Passed    0.02 sec
          Start   12: chest_test_dl_cellid2_50prb
  12/1523 Test   #12: chest_test_dl_cellid2_50prb .....................................   Passed    0.01 sec
          Start   13: chest_test_ul_cellid0
  13/1523 Test   #13: chest_test_ul_cellid0 ...........................................   Passed    0.04 sec
          Start   14: chest_test_ul_cellid1
  14/1523 Test   #14: chest_test_ul_cellid1 ...........................................   Passed    0.04 sec
          Start   15: chest_test_ul_cellid2
  15/1523 Test   #15: chest_test_ul_cellid2 ...........................................   Passed    0.04 sec
          Start   16: chest_test_srs_6
  16/1523 Test   #16: chest_test_srs_6 ................................................   Passed    1.25 sec
          Start   17: chest_test_srs_15
  17/1523 Test   #17: chest_test_srs_15 ...............................................   Passed    1.26 sec
          Start   18: chest_test_srs_25
  18/1523 Test   #18: chest_test_srs_25 ...............................................   Passed    1.38 sec
          Start   19: chest_test_srs_50
  19/1523 Test   #19: chest_test_srs_50 ...............................................   Passed    1.31 sec
          Start   20: chest_test_srs_75
  20/1523 Test   #20: chest_test_srs_75 ...............................................   Passed    1.43 sec
          Start   21: chest_test_srs_100
  21/1523 Test   #21: chest_test_srs_100 ..............................................   Passed    1.55 sec
          Start   22: chest_nbiot_test_dl
  22/1523 Test   #22: chest_nbiot_test_dl .............................................   Passed    0.00 sec
          Start   23: chest_test_sl_psbch
  23/1523 Test   #23: chest_test_sl_psbch .............................................   Passed    0.00 sec
          Start   24: dmrs_pdsch_test
  24/1523 Test   #24: dmrs_pdsch_test .................................................   Passed    3.55 sec
          Start   25: dmrs_pdcch_test_non_interleaved
  25/1523 Test   #25: dmrs_pdcch_test_non_interleaved .................................   Passed    2.07 sec
          Start   26: dmrs_pdcch_test_interleaved
  26/1523 Test   #26: dmrs_pdcch_test_interleaved .....................................   Passed    1.48 sec
          Start   27: csi_rs_test
  27/1523 Test   #27: csi_rs_test .....................................................   Passed    0.00 sec
          Start   28: csi_rs_pattern_test
  28/1523 Test   #28: csi_rs_pattern_test .............................................   Passed    0.00 sec
          Start   29: sequence_test
  29/1523 Test   #29: sequence_test ...................................................   Passed    0.01 sec
          Start   30: sliv_test_14
  30/1523 Test   #30: sliv_test_14 ....................................................   Passed    0.00 sec
          Start   31: sliv_test_52
  31/1523 Test   #31: sliv_test_52 ....................................................   Passed    0.00 sec
          Start   32: phy_common_test
  32/1523 Test   #32: phy_common_test .................................................   Passed    0.00 sec
          Start   33: block_test_20
[... 151 lines cut ...]
          Start  109: pscch_test_tm2_p6
 109/1523 Test  #109: pscch_test_tm2_p6 ...............................................   Passed    0.00 sec
          Start  110: pscch_test_tm2_p15
 110/1523 Test  #110: pscch_test_tm2_p15 ..............................................   Passed    0.00 sec
          Start  111: pscch_test_tm2_p25
 111/1523 Test  #111: pscch_test_tm2_p25 ..............................................   Passed    0.00 sec
          Start  112: pscch_test_tm2_p50
 112/1523 Test  #112: pscch_test_tm2_p50 ..............................................   Passed    0.00 sec
          Start  113: pscch_test_tm2_p75
 113/1523 Test  #113: pscch_test_tm2_p75 ..............................................   Passed    0.00 sec
          Start  114: pscch_test_tm2_p100
 114/1523 Test  #114: pscch_test_tm2_p100 .............................................   Passed    0.00 sec
          Start  115: pscch_test_tm4_p6
 115/1523 Test  #115: pscch_test_tm4_p6 ...............................................   Passed    0.00 sec
          Start  116: pscch_test_tm4_p15
 116/1523 Test  #116: pscch_test_tm4_p15 ..............................................   Passed    0.00 sec
          Start  117: pscch_test_tm4_p25
 117/1523 Test  #117: pscch_test_tm4_p25 ..............................................   Passed    0.00 sec
          Start  118: pscch_test_tm4_p50
 118/1523 Test  #118: pscch_test_tm4_p50 ..............................................   Passed    0.00 sec
          Start  119: pscch_test_tm4_p75
 119/1523 Test  #119: pscch_test_tm4_p75 ..............................................   Passed    0.00 sec
          Start  120: pscch_test_tm4_p100
 120/1523 Test  #120: pscch_test_tm4_p100 .............................................   Passed    0.00 sec
          Start  121: pssch_test_tm2_p6

closed 2024-01-17 by reallsj

tilldroemmer, 2024-01-23:
@reallsj I noticed you closed the issue, how did you solve it? I am stuck at the same test

reallsj, 2024-01-23:
> @reallsj I noticed you closed the issue, how did you solve it? I am stuck at the same test

Just wait it to move on, dont close the terminal.