srsran/srsRAN_4G #1261: How pack a ue alert in asn1::rrc::paging_record_s paging_record; in srsran project

stackprogramer, 2023-12-22:
I asked my question on support mailing list but No one answered, so I asked it here...
The asn1::rrc::paging_record_s paging_record is a structure used in the
srsRAN project to handle paging records. However, the specific details on
how to pack a UE alert in this structure are not directly available in the
search results.

Till mobiles that connect my srsran enb they see a ue alert.

more info about  eu-alert

https://en.wikipedia.org/wiki/EU-Alert

 srsRAN community or the maintainers of the project can guide me in the
implementation ue-alert in srsran enb 4g project.
Thanks in advance

andrepuschmann, 2023-12-22:
Hey - adding support for CMAS is certainly possible and has actually been done already. Those modifications are not public domain as far as I know. From a code perspective the main challenge is to rewrite parts of the MAC code to allow dynamic recreation of SIB messages. Currently they are generate upon start of the eNB and do not change during the runtime.

stackprogramer, 2023-12-24:
I saw that this issue (https://github.com/srsran/srsRAN_Project/discussions/53    ) but when I tested it, In sib file some section is changed  si_mapping_info = [ 3,12 ];
It adds si_mapping_info new index 12, and sib 12 definition.....
I did this solution but my enb can not detect it after changing.....But Changing Mac layer for paging is not mentioned in it....

```
sib12 =
{
data_coding_scheme_r9_present = true;
msg_id_r9 = 4370;
serial_num_r9 = 3007;
warning_msg_segment_type =
{
warning_msg_segment_type_r9 = "lastSegment";
};
data_coding_scheme_r9 = "01";
warning_msg_segment_r9 = "12345678901234567890123456789012345678901234567890123456789012345678901234567890129";
};

```