srsran/srsRAN_4G #1273: Raspberry Pi5 Build Failure w/ LimeSDR Mini

perfectionst, 2024-01-22:
## Issue Description ##
Unable to build from source on Raspberry Pi 5 with Ubuntu 23.10 (22.04 LTS not supported) with a LimeSDR Mini 2.

## Setup Details ##
Raspberry Pi5 8GB
Ubuntu 23.10
LimeSDR Mini 2
gcc 11.4.0 (gcc 12 and gcc 9 also failed)
Latest srsRAN release (23.11) as well as 23.04

## Expected Behavior ##
Successful Build

## Actual Behaviour ##
2 Errors based on memmove in stl_algobase.h:437:30:

[ 51%] Building CXX object lib/test/asn1/CMakeFiles/srsran_asn1_rrc_meas_test.dir/srsran_asn1_rrc_meas_test.cc.o
In file included from /usr/include/c++/13/bits/stl_uninitialized.h:63,
                 from /usr/include/c++/13/memory:69,
                 from /home/test/srsRAN_4G/lib/include/srsran/common/common.h:31,
                 from /home/test/srsRAN_4G/lib/include/srsran/common/byte_buffer.h:25,
                 from /home/test/srsRAN_4G/lib/include/srsran/common/buffer_pool.h:25,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/asn1_utils.h:25,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/rrc/common.h:31,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/rrc/rr_common.h:31,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/rrc/common_ext.h:31,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/rrc/meascfg.h:31,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/rrc/bcch_msg.h:31,
                 from /home/test/srsRAN_4G/lib/include/srsran/asn1/rrc/dl_dcch_msg.h:31,
                 from /home/test/srsRAN_4G/lib/test/asn1/srsran_asn1_rrc_meas_test.cc:22:
In static member function ‘static _Up* std::__copy_move<_IsMove, true, std::random_access_iterator_tag>::__copy_m(_Tp*, _Tp*, _Up*) [with _Tp = asn1::rrc::meas_id_to_add_mod_s; _Up = asn1::rrc::meas_id_to_add_mod_s; bool _IsMove = false]’,
    inlined from ‘_OI std::__copy_move_a2(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::meas_id_to_add_mod_s*; _OI = asn1::rrc::meas_id_to_add_mod_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:506:30,
    inlined from ‘_OI std::__copy_move_a1(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::meas_id_to_add_mod_s*; _OI = asn1::rrc::meas_id_to_add_mod_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:533:42,
    inlined from ‘_OI std::__copy_move_a(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::meas_id_to_add_mod_s*; _OI = asn1::rrc::meas_id_to_add_mod_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:540:31,
    inlined from ‘_OI std::copy(_II, _II, _OI) [with _II = asn1::rrc::meas_id_to_add_mod_s*; _OI = asn1::rrc::meas_id_to_add_mod_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:633:7,
    inlined from ‘void asn1::dyn_array<T>::resize(uint32_t, uint32_t) [with T = asn1::rrc::meas_id_to_add_mod_s]’ at /home/test/srsRAN_4G/lib/include/srsran/asn1/asn1_utils.h:248:18,
    inlined from ‘int test_meas_config()’ at /home/test/srsRAN_4G/lib/test/asn1/srsran_asn1_rrc_meas_test.cc:183:42:
/usr/include/c++/13/bits/stl_algobase.h:437:30: error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ forming offset 3 is out of the bounds [0, 3] [-Werror=array-bounds=]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In static member function ‘static _Up* std::__copy_move<_IsMove, true, std::random_access_iterator_tag>::__copy_m(_Tp*, _Tp*, _Up*) [with _Tp = asn1::rrc::quant_cfg_nr_r15_s; _Up = asn1::rrc::quant_cfg_nr_r15_s; bool _IsMove = false]’,
    inlined from ‘_OI std::__copy_move_a2(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:506:30,
    inlined from ‘_OI std::__copy_move_a1(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:533:42,
    inlined from ‘_OI std::__copy_move_a(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:540:31,
    inlined from ‘_OI std::copy(_II, _II, _OI) [with _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:633:7,
    inlined from ‘void asn1::dyn_array<T>::resize(uint32_t, uint32_t) [with T = asn1::rrc::quant_cfg_nr_r15_s]’ at /home/test/srsRAN_4G/lib/include/srsran/asn1/asn1_utils.h:248:18,
    inlined from ‘int test_meas_config()’ at /home/test/srsRAN_4G/lib/test/asn1/srsran_asn1_rrc_meas_test.cc:194:57:
/usr/include/c++/13/bits/stl_algobase.h:437:30: error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ forming offset 36 is out of the bounds [0, 36] [-Werror=array-bounds=]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

## Steps to reproduce the problem ##
Attempt to build on a Raspberry Pi 5

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

[CMAKE SYSTEM Linux 6 5.txt](https://github.com/srsran/srsRAN_4G/files/14006092/CMAKE.SYSTEM.Linux.6.5.txt)

chrboesch, 2024-02-13:
Get the same error on AMD64.

andrepuschmann, 2024-05-27:
The newer Ubuntu/gcc versions are not used in the CI I am afraid. But it seems that Raspberry Pi OS does not have those build issues and I can also recommend Raspberry Pi OS over Ubuntu on the Pi as it's also easier to get the eNB running because the SCTP kernel module (which is needed to connect to the 4G core) is already built-in.