srsran/srsRAN_4G #1280: enb Tx port not specified in  5G SA End-to-End

chakriganap, 2024-01-30:
## Issue Description ##
Unable to connect the eNode B for setup 5G SA End-to-End configuration. But its working fine with  srsRAN 4G with ZMQ Virtual Radios

## Setup Details ##
Ubuntu 20.04.6 LTS, Linux 5.4.0-169-generic, VM, srsRAN 4G with ZMQ Virtual Radios and Free5GC Core
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
connect eNodeB

## Actual Behaviour ##
[zmq] enb Tx port not specified. Disabling transmitter.
[zmq] enb Rx port not specified. Disabling receiver.
[zmq] Error: Neither Tx port nor Rx port specified.
Error initializing radio.

## Steps to reproduce the problem ##
run the enb config file 

## Additional Information ##

chakriganap, 2024-01-30:
If I remove the nr_cell_list data in rr.config file. the eNodeB connected. 

Thanks

joaquim-srs, 2024-03-14:
Hi there,

As you already figured out, this may happen in situations where the ENB/GNB configuration requires more antenna ports than the number of ZMQ ports provided in the RF arguments. You can solve this by adding additional ZMQ ports to the `device_args`, for example:

```
device_args = fail_on_disconnect=true,id=enb,base_srate=23.04e6,tx_port0=tcp://*:5000,rx_port0=tcp://localhost:5100,tx_port1=tcp://*:5001,rx_port1=tcp://localhost:5101. . .
```