srsran/srsRAN_4G #1288: srsRAN gnb cannot find USRP B210

bcgordon3, 2024-02-21:
## Issue Description ##
I have a new build on Ubuntu 22.04. I have seen similar issues, but not the same. This one seems like I am just missing a fundamental step somewhere but can't figure out where. The srsRAN gNB SW cannot find the UHD device (USRP B210) connected to the system. I get the following error when trying to start:

No radio devices available.
srsGNB ERROR: Unable to create radio session.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
I have the latest source (commit 50fe9623c) as of 2/21/2024.
Ubuntu 22.04.03
Did the UHD repo addition and install:
   sudo add-apt-repository ppa:ettusresearch/uhd
   sudo apt-get update
   sudo apt-get install libuhd-dev uhd-host

B210 connected via USB 3.1 to PC with intel i7-1185G7E, 16GB RAM

## Expected Behavior ##
I expected the gnb SW to run and start transmitting.

## Actual Behaviour ##
Here is the output I get when changing the AMF IP in the config file to my local 5G core:

sudo gnb -c gnb_rf_b210_fdd_srsUE.yml

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in quad executor mode.

--== srsRAN gNB (commit 50fe9623c) ==--

Connecting to AMF on 10.10.4.1:38412
No radio devices available.
srsGNB ERROR: Unable to create radio session.

## Steps to reproduce the problem ##
Only modified line in the config is the AMF and bind IP addresses.

## Additional Information ##
I did a make clean, git pull, then redid the whole build and test process with success. The UHD device is found on the PC. Below are the cmake output and the summary lines from make then make tests. Following that is the output from the uhd_find_devices and uhd_usrp_probe.

cmake output:

cmake ../
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release.
-- Performing Test HAS_MAYBE_UNINITIALIZED
-- Performing Test HAS_MAYBE_UNINITIALIZED - Success
-- Performing Test HAVE_NON_VIRTUAL_DTOR
-- Performing Test HAVE_NON_VIRTUAL_DTOR - Success
-- Performing Test HAVE_SUGGEST_OVERRIDE
-- Performing Test HAVE_SUGGEST_OVERRIDE - Success
-- Performing Test HAVE_SHADOW
-- Performing Test HAVE_SHADOW - Success
-- Assertion level set to NORMAL
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR)
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR)
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR)
-- Found Backward: /home/siguser/srsgnb/srsRAN_Project/cmake/modules
-- Backward-cpp found, but external libraries are missing.
[... 166 lines cut ...]
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX2
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

bcgordon3, 2024-02-27:
I guess nobody else has had this issue. Bump for MWC.

bcgordon3, 2024-03-15:
Final comment. I had to reinstall UHD drivers then rebuild srsRAN and then it started working.

closed 2024-03-15 by bcgordon3

sambak22, 2024-04-02:
Hello! 
I have successfully built srsRAN 5G project in Ubuntu 22.04. Further, I have also connected USRP B205 Mini. When I am running ./gnB compiler using B200 TDD config file, I am unable to get following output:
Available radio types: uhd.

--== srsRAN gNB (commit srsran/srsRAN_Project@77be7d3) ==--

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.HEAD-0-g197cdc4f
Making USRP object with args 'type=b200'
Cell pci=1, bw=10 MHz, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

==== gNodeB started ===
Type to view trace

Please guide me to get through this step. Also, please assist me in setting up 5G SA. I have successfully built open5gs as core network and tested it using UERANSIM. I would want to run open5gs and srsRAN in same machine installed with Ubuntu 22.04.

Thank you
Regards 

P.S.: I am having similar issue as mentioned in your post. I followed your final comment by re-installing UHD drivers but no expected output is being generated as mentioned above.

bcgordon3, 2024-04-02:
I can't guide you here but maybe can point to a couple things I see.
1) You are not connecting to an AMF (5G Core) -- the filed SCTP tells you that. You need to fix that.
2) You have a channel spacing of 30kHz in N78 on 10MHz. Pretty sure the 10 MHz will only support 15KHz spacing.

Good luck.