srsran/srsRAN_4G #1295: Fails to build on Debian 13 (trixie/testing)

kotarou3, 2024-03-03:
## Issue Description ##
On a fresh Debian 13 installation, srsRAN fails to build from source, failing with one of two kinds of errors:

(if UHD deps are installed)
```
-- Could NOT find Boost: missing: system (found /usr/lib/x86_64-linux-gnu/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.35"))
```

(if UHD deps aren't installed or disabled with `-DENABLE_UHD=no`)
```
In file included from /srsRAN_4G/srsue/hdr/phy/phy_metrics.h:26,
                 from /srsRAN_4G/srsue/hdr/phy/phy_common.h:25,
                 from /srsRAN_4G/srsue/src/phy/phy_common.cc:26:
In member function 'std::array<_Tp, _Nm>::value_type& std::array<_Tp, _Nm>::operator[](size_type) [with _Tp = float; long unsigned int _Nm = 5]',
    inlined from 'void srsue::phy_common::reset_measurements(uint32_t)' at /srsRAN_4G/srsue/src/phy/phy_common.cc:662:18,
    inlined from 'void srsue::phy_common::reset()' at /srsRAN_4G/srsue/src/phy/phy_common.cc:918:21:
/usr/include/c++/13/array:203:24: error: array subscript 5 is above array bounds of 'std::__array_traits<float, 5>::_Type' {aka 'float [5]'} [-Werror=array-bounds=]
  203 |         return _M_elems[__n];
      |                ~~~~~~~~^
```

srsRAN builds fine on Debian 12, so it might be due to new compiler and boost versions with Debian 13:
<table>
<tr><th>Tool</th><th>Debian 12</th><th>Debian 13</th></tr>
<tr><td>GCC</td><td>12.2.0</td><td>13.2.0</td></tr>
<tr><td>Boost</td><td>1.74.0</td><td>1.83.0</td></tr>
</table>

## Setup Details ##
- [Installed package list `dpkg -l`](https://github.com/srsran/srsRAN_4G/files/14471944/dpkg-l.log)
- srsRAN commit ec29b0c1ff79cebcbe66caa6d6b90778261c42b8
- Fresh clone from Github and build

## Expected Behavior ##
srsRAN builds successfully

## Actual Behaviour ##
srsRAN fails to build

## Steps to reproduce the problem ##
1. Run Debian 13 (e.g., by booting from a [live CD](https://cdimage.debian.org/cdimage/weekly-live-builds/amd64/iso-hybrid/))
2. Ensure packages are up to date: `sudo apt update && sudo apt upgrade`
3. Install git `sudo apt install git`
4. Install srsRAN dependencies: `sudo apt install build-essential cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev`
5. Install optional dependencies: `sudo apt install libsoapysdr-dev libbladerf-dev libuhd-dev libzmq3-dev uhd-host --no-install-recommends`
6. Clone and build srsRAN: `git clone https://github.com/srsran/srsRAN_4G.git && mkdir srsRAN_4G/build && cd srsRAN_4G/build && cmake ..`
7. Try again without UHD: `cmake .. -DENABLE_UHD=no && make -j`

kotarou3, 2024-03-03:
I can work around this issue if I disable both UHD and -Werror with:
```
cmake .. -DENABLE_UHD=no -DENABLE_WERROR=no
```

pedrib, 2024-06-09:
Had the same problem, to fix it in the cmake step:
`cmake .. -DCMAKE_CXX_FLAGS="-Wno-array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread"`