srsran/srsRAN_4G #1299: decoding using npdsch_ue failed

ahmedelbushra, 2024-03-12:
## Issue Description ##
I am using RTL-SDR to detect and decode NB-IoT, I was able to detect but not decode it.

## Setup Details ##
step1: run ./cell_search_nbiot -b 20 
[image]

step2: ./npdsch_ue -f 815.9e6 
[image]

## Expected Behavior ##
[image]

## Actual Behavior ##
[image]

andrepuschmann, 2024-03-12:
The cell ID detection might not work super reliably depending on your RF situation. Have you verified there is an NB-IoT carrier at all in on that frequency? You can do that with any spectrum monitoring app.

ahmedelbushra, 2024-03-13:
I am RTL SDR Gqrx to view the spectrum : the cell_search_nbiot  detected 2 cells,
[image]
The output of the Gqrx : 
[image]
It show no signals in that range.

andrepuschmann, 2024-03-13:
Hard to say if there is an NB-IoT carrier at all. The NPSS power is very low, too low to reliably sync on it anyway. Maybe try to get closer to the cell. See [here](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/nbiot/source/index.html) how the spectrum should look like.

closed 2024-03-13 by andrepuschmann