srsran/srsRAN_4G #1303: Error running make test

raul-parada, 2024-03-18:
When running make test from the latest version, it didn't pass the whole suite of tests:

[image]

I'm using this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#multi-ue-emulation

Distributor ID: Ubuntu
Description: Ubuntu 22.04.4 LTS
Release: 22.04
Codename: jammy

pgawlowicz, 2024-03-22:
Could you provide gnb and srsUE logs?

raul-parada, 2024-03-22:
Please, tell me how to get them.

pgawlowicz, 2024-03-22:
They are located at `/tmp/gnb.log`, and `/tmp.ueX.log`

Could you try running this command in the srsRAN-4G build directory:
```
./lib/src/radio/test/benchmark_radio -d zmq -a "tx_port=tcp://*:2000,rx_port=tcp://localhost:2000,tx_port=tcp://*:2001,rx_port=tcp://localhost:2001,tx_port=tcp://*:2002,rx_port=tcp://localhost:2002,tx_port=tcp://*:2003,rx_port=tcp://localhost:2003" -p 4 -g 100
```

and post its output here.

raul-parada, 2024-03-22:
> They are located at `/tmp/gnb.log`, and `/tmp.ueX.log`
>
> Could you try running this command in the srsRAN-4G build directory:
>
> ```
> ./lib/src/radio/test/benchmark_radio -d zmq -a [...]

Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins:
Instantiating objects and allocating memory...
Initialising instances...
Opening 4 channels in RF device=zmq with args=tx_port=tcp://*:2000,rx_port=tcp:/                               /localhost:2000,tx_port=tcp://*:2001,rx_port=tcp://localhost:2001,tx_port=tcp://                               *:2002,rx_port=tcp://localhost:2002,tx_port=tcp://*:2003,rx_port=tcp://localhost                               :2003
Supported RF device list: zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2000
Error: binding transmitter socket (tcp://*:2000): Address already in use
[zmq] Error: opening transmitter
Error: Calling radio_multi constructor
Tearing down...
Failed!
[gnb.log](https://github.com/srsran/srsRAN_4G/files/14720683/gnb.log)
[ue.log](https://github.com/srsran/srsRAN_4G/files/14720684/ue.log)

pgawlowicz, 2024-03-22:
It seems that some active process is already using the socket. Please kill the process and try again. You can find out the process ID with:
```
sudo netstat -nlp | grep 2000
```

raul-parada, 2024-03-22:
> It seems that some active process is already using the socket. Please kill the process and try again. You can find out the process ID with:
> 
> ```
> sudo netstat -nlp | grep 2000
> ```

is it possible to change the port of gNB (2000) for another one? another required software is using it.

pgawlowicz, 2024-03-22:
Sure, you can set any port you want. You have to change the port number in the srsgnb config and also in gnuradio-compantion.