srsran/srsRAN_4G #1304: AMF report ERROR:  HTTP response error with code [400]

achkonsa, 2024-03-19:
## Issue Description ##
[When i start to run the UE, the amf terminal always report the ERROR: "HTTP response error [400]" and registration reject next line, then the UE terminal reported "Received RRC Release" .]

## Setup Details ##
[First, start the open5gs
Second, start the gnb from srsRAN_Project
Finally, start the NR UE from srsRAN_4G]

## Expected Behavior ##
[10/11 20:35:48.444: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
10/11 20:35:48.444: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2523)
10/11 20:35:48.444: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[1] CellID[0x19b0] (../src/amf/ngap-handler.c:562)
10/11 20:35:48.444: [amf] INFO: [suci-0-001-01-0000-0-0-0000000000] Unknown UE by SUCI (../src/amf/context.c:1789)
10/11 20:35:48.444: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1570)
10/11 20:35:48.444: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1061)
10/11 20:35:48.444: [gmm] INFO: [suci-0-001-01-0000-0-0-0000000000]    SUCI (../src/amf/gmm-handler.c:157)
10/11 20:35:48.447: [sbi] WARNING: [2e5782e8-682a-41ee-a705-499077b49eb3] (NF-discover) NF has already been added (../lib/sbi/nnrf-handler.c:833)
10/11 20:35:48.447: [sbi] WARNING: NF EndPoint updated [127.0.0.12:80] (../lib/sbi/context.c:1623)
10/11 20:35:48.447: [sbi] WARNING: NF EndPoint updated [127.0.0.12:7777] (../lib/sbi/context.c:1532)
10/11 20:35:48.447: [sbi] WARNING: NF EndPoint updated [127.0.0.12:7777] (../lib/sbi/context.c:1532)
10/11 20:35:48.447: [sbi] WARNING: NF EndPoint updated [127.0.0.12:7777] (../lib/sbi/context.c:1532)
10/11 20:35:48.448: [sbi] INFO: [2e5782e8-682a-41ee-a705-499077b49eb3] (NF-discover) NF Profile updated (../lib/sbi/nnrf-handler.c:856)
10/11 20:35:48.598: [sbi] WARNING: [2e5e0190-682a-41ee-8e27-abb8e1c20f56] (NF-discover) NF has already been added (../lib/sbi/nnrf-handler.c:833)
10/11 20:35:48.598: [sbi] WARNING: NF EndPoint updated [127.0.0.20:80] (../lib/sbi/context.c:1623)
10/11 20:35:48.598: [sbi] WARNING: NF EndPoint updated [127.0.0.20:7777] (../lib/sbi/context.c:1532)
10/11 20:35:48.598: [sbi] INFO: [2e5e0190-682a-41ee-8e27-abb8e1c20f56] (NF-discover) NF Profile updated (../lib/sbi/nnrf-handler.c:856)
10/11 20:35:48.784: [gmm] INFO: [imsi-001010000000000] Registration complete (../src/amf/gmm-sm.c:1993)]

## Actual Behaviour ##
[03/15 16:33:57.996: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
03/15 16:33:57.997: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2556)
03/15 16:33:57.997: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[1] CellID[0x19b0] (../src/amf/ngap-handler.c:562)
03/15 16:33:57.997: [amf] INFO: [suci-0-001-01-0000-0-0-0000000000] Unknown UE by SUCI (../src/amf/context.c:1839)
03/15 16:33:57.997: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1620)
03/15 16:33:57.997: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
03/15 16:33:57.997: [gmm] INFO: [suci-0-001-01-0000-0-0-0000000000]    SUCI (../src/amf/gmm-handler.c:172)
03/15 16:33:57.999: [gmm] ERROR: [suci-0-001-01-0000-0-0-0000000000] HTTP response error [400] (../src/amf/gmm-sm.c:1715)
03/15 16:33:57.999: [amf] WARNING: [suci-0-001-01-0000-0-0-0000000000] Registration reject [95] (../src/amf/nas-path.c:206)
03/15 16:33:58.217: [amf] INFO: UE Context Release [Action:3] (../src/amf/ngap-handler.c:1696)
03/15 16:33:58.217: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1697)
03/15 16:33:58.217: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0000000000] (../src/amf/ngap-handler.c:1701)
03/15 16:33:58.217: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2563)
03/15 16:33:58.217: [amf] INFO: [Removed] Number of AMF-UEs is now 0 (../src/amf/context.c:1713)]

## Steps to reproduce the problem ##
[First, start the open5gs
Second, start the gnb from srsRAN_Project
Finally, start the NR UE from srsRAN_4G]

## Additional Information ##
[when meet this error, the UE will report "Received RRC Release." And I have registered subscriber information even though the AMF reported "[amf] INFO: [suci-0-001-01-0000-0-0-0000000000] Unknown UE by SUCI" ]
[image]

achkonsa, 2024-04-19:
could somebody help me T_T

herlesupreeth, 2024-04-21:
> 03/15 16:33:57.999: [amf] WARNING: [suci-0-001-01-0000-0-0-0000000000] Registration reject [95] 

The above error suggests that you have not provisioned the UE SIM details correctly in open5gs Web UI

MrPluto0, 2024-04-22:
> > 03/15 16:33:57.999: [amf] WARNING: [suci-0-001-01-0000-0-0-0000000000] Registration reject [95]
> 
> The above error suggests that you have not provisioned the UE SIM details correctly in open5gs Web UI

hi, my SIM info has uploaded to open5gs, which can been seen as follows. But the question still exist. I paste the log of open5gs-amf when running gnb and ue.

[image]

```
4月 22 15:52:57 open5gs-vm open5gs-amfd[5511]: 04/22 15:52:57.534: [amf] INFO: gNB-N2 accepted[127.0.0.1]:47108 in ng-path module (../src/amf/ngap-sctp.c:113)
4月 22 15:52:57 open5gs-vm open5gs-amfd[5511]: 04/22 15:52:57.534: [amf] INFO: gNB-N2 accepted[127.0.0.1] in master_sm module (../src/amf/amf-sm.c:757)
4月 22 15:52:57 open5gs-vm open5gs-amfd[5511]: 04/22 15:52:57.545: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1236)
4月 22 15:52:57 open5gs-vm open5gs-amfd[5511]: 04/22 15:52:57.545: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:796)
4月 22 16:06:17 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:17.388: [amf] INFO: gNB-N2[127.0.0.1] connection refused!!! (../src/amf/amf-sm.c:809)
4月 22 16:06:17 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:17.389: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1263)
4月 22 16:06:20 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:20.323: [amf] INFO: gNB-N2 accepted[127.0.0.1]:60156 in ng-path module (../src/amf/ngap-sctp.c:113)
4月 22 16:06:20 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:20.323: [amf] INFO: gNB-N2 accepted[127.0.0.1] in master_sm module (../src/amf/amf-sm.c:757)
4月 22 16:06:20 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:20.331: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1236)
4月 22 16:06:20 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:20.331: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:796)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.927: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.927: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2662)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.927: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.927: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] Unknown UE by SUCI (../src/amf/context.c:1844)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.928: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1621)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.928: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1214)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.928: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:172)
4月 22 16:06:33 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:33.937: [sbi] INFO: [AUSF] (SCP-discover) NF registered [9e636f3a-007c-41ef-9ec5-a93f9e97a1a8:1] (../lib/sbi/path.c:211)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.026: [sbi] INFO: [UDM] (SCP-discover) NF registered [9e64e946-007c-41ef-a2c6-15a46ab24f96:1] (../lib/sbi/path.c:211)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.031: [sbi] WARNING: [UDM] (SCP-discover) NF has already been added [9e64e946-007c-41ef-a2c6-15a46ab24f96:2] (../lib/sbi/path.c:216)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.042: [amf] WARNING: NF EndPoint(addr) updated [127.0.0.12:7777] (../src/amf/npcf-handler.c:143)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.199: [gmm] INFO: [imsi-001010123456780] Registration complete (../src/amf/gmm-sm.c:2313)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [amf] INFO: [imsi-001010123456780] Configuration update command (../src/amf/nas-path.c:591)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [gmm] INFO:     UTC [2024-04-22T08:06:34] Timezone[0]/DST[0] (../src/amf/gmm-build.c:558)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [gmm] INFO:     LOCAL [2024-04-22T16:06:34] Timezone[28800]/DST[0] (../src/amf/gmm-build.c:563)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [amf] INFO: [Added] Number of AMF-Sessions is now 1 (../src/amf/context.c:2683)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [gmm] WARNING: [imsi-001010123456780] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1272)
4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.231: [gmm] INFO: [imsi-001010123456780] No GUTI allocated (../src/amf/gmm-sm.c:1591)
4月 22 16:06:37 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:37.619: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1696)
4月 22 16:06:37 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:37.619: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1697)
4月 22 16:06:37 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:37.619: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0123456780] (../src/amf/ngap-handler.c:1701)
4月 22 16:06:37 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:37.619: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2669)
```

achkonsa, 2024-04-22:
Maybe the apn should be "internet", instead of "srspan"？ And maybe you forgot to add [slicing] information. 
It`s my ue.conf:
`[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = E8ED289DEBA952E4283B54E88E6183CA
k    = 465B5CE8B199B49FAA5F0A2EE238A6BC
imsi = 001010000000000
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[slicing]
enable = true
nssai-sst = 1
nssai-sd = 000001

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false`

achkonsa, 2024-04-22:
Maybe the apn should be "internet", instead of "srspan"？ And maybe you forgot to add [slicing] information.
Its my ue.conf: 
`[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc = E8ED289DEBA952E4283B54E88E6183CA
k = 465B5CE8B199B49FAA5F0A2EE238A6BC
imsi = 001010000000000
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[slicing]
enable = true
nssai-sst = 1
nssai-sd = 000001

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false`

achkonsa, 2024-04-22:
Maybe the apn should be "internet", instead of "srspan"？ And maybe you forgot to add [slicing] information.
Its my ue.conf:
`
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = E8ED289DEBA952E4283B54E88E6183CA
k    = 465B5CE8B199B49FAA5F0A2EE238A6BC
imsi = 001010000000000
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[slicing]
enable = true
nssai-sst = 1
nssai-sd = 000001

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false
`

MrPluto0, 2024-04-22:
> Maybe the apn should be "internet", instead of "srspan"？ And maybe you forgot to add [slicing] information. Its my ue.conf: `[rf] freq_offset = 0 tx_gain = 50 rx_gain = 40 srate = 23.04e6 [...]

thanks, but it doesn't work. But i use the open5gs-docker in `srsRAN_Project`, it can work properly.

so i just want to verify if it can been done by installing open5gs locally but not docker? becase of the poor network from china and complicate docker env.  @herlesupreeth 

i think manually install can help me to understand the overrall structures.

achkonsa, 2024-04-22:
> > Maybe the apn should be "internet", instead of "srspan"？ And maybe you forgot to add [slicing] information. Its my ue.conf: `[rf] freq_offset = 0 tx_gain = 50 rx_gain = 40 srate = 23.04e6 nof_antennas = 1
> > device_name = zmq device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6
> > [rat.eutra] dl_earfcn = 2850 nof_carriers = 0
> > [rat.nr] bands = 3 nof_carriers = 1 max_nof_prb = 106 nof_prb = 106
> > [pcap] enable = none mac_filename = /tmp/ue_mac.pcap mac_nr_filename = /tmp/ue_mac_nr.pcap nas_filename = /tmp/ue_nas.pcap
> > [log] all_level = info phy_lib_level = none all_hex_limit = 32 filename = /tmp/ue.log file_max_size = -1
> > [usim] mode = soft algo = milenage opc = E8ED289DEBA952E4283B54E88E6183CA k = 465B5CE8B199B49FAA5F0A2EE238A6BC imsi = 001010000000000 imei = 353490069873319
> > [rrc] release = 15 ue_category = 4
> > [nas] apn = internet apn_protocol = ipv4
> > [slicing] enable = true nssai-sst = 1 nssai-sd = 000001
> > [gw] netns = ue1 ip_devname = tun_srsue ip_netmask = 255.255.255.0
> > [gui] enable = false`
> 
> thanks, but it doesn't work. But i use the open5gs-docker in `srsRAN_Project`, it can work properly.
> 
> so i just want to verify if it can been done by installing open5gs locally but not docker? @herlesupreeth
> 
> not only becase of the poor network from china and complicate docker env, but manual install can help understand the structures.

可以的，我用的就是单独安装的open5gs

herlesupreeth, 2024-04-22:
> 4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [gmm] WARNING: [imsi-001010123456780] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1272)

@MrPluto0 you have to have `srsapn` as the DNN/APN in the open5gs WebUI

MrPluto0, 2024-04-22:
> > 4月 22 16:06:34 open5gs-vm open5gs-amfd[5511]: 04/22 16:06:34.200: [gmm] WARNING: [imsi-001010123456780] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1272)
> 
> @MrPluto0 you have to have `srsapn` as the DNN/APN in the open5gs WebUI

oh thanks, it works!
I thought it justs need the same usim, but don't know the apn

herlesupreeth, 2024-04-22:
Nice.. i will close this issue then