srsran/srsRAN_4G #1311: Couldn't stop after 5s. Forcing exit: Why UE in srsran4g can not stop after some time that i run.....it will be killed (UE process)?

stackprogramer, 2024-03-29:
**Couldn't stop after 5s. Forcing exit:** Why UE in srsran4g can not stop after some time that I run it....for stopping it will be killed (UE process)?
It attempts to login to an ENB and .....
I don't have any problem in stopping ENB or EPC but for a UE in srsan4g project, I see that stopping has some bug.
I report this bug.

Could not find Home PLMN Id=00101, trying to connect to PLMN Id=00101
RF status: O=13, U=0, L=0
.RF status: O=11, U=0, L=0
^CStopping ..
stack layer is stoped!
gw layer is stoped!
Couldn't stop after 5s. Forcing exit.
Saving NAS PCAP file (DLT=148) to  
Killed

Raziahmanzoor, 2024-07-12:
did you solve this issue? i am facing the same problem

stackprogramer, 2024-07-13:
@Raziahmanzoor Hi, I'm sorry. I didn't find any solution.