srsran/srsRAN_4G #1316: configuration when ue and gnb installed in separate machines

faratell, 2024-04-05:
Hello.
I have installed open5gs in one machine with this ip: 172.20.70.126
, srsRAN_project in another machine as zmq gnb : 172.20.70.110
and srsRAN_4G in another machine as zmq ue: 172.20.70.115
open5gs and gnb run correctly and gnb successfully connected to amf.
in gnb_zmq.yaml Iset 
device_args: tx_port=tcp://172.20.70.110:2000,rx_port=tcp://172.20.70.110:200>

in ue_zmq.yaml I set:

device_name = zmq
device_args = tx_port=tcp://172.20.70.115:2001,rx_port=tcp://172.20.70.115:2000,base_srate=23.04e6

where shoulde be set the ip address of gnb for ue and how recognized it?
## Issue Description ##
 ./srsue ./ue_zmq.yaml
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins:
Reading configuration file ./ue_zmq.yaml...

Built in Release mode using commit ec29b0c1f on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://172.20.70.115:2001,rx_port=tcp://172.20.70.115:2000,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://172.20.70.115:2000
CH0 tx_port=tcp://172.20.70.115:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...

s5uishida, 2024-04-07:
@faratell 

Is the following helpful?

https://github.com/s5uishida/open5gs_5gc_srsran_sample_config

faratell, 2024-04-07:
Thank you very much for your advice. I will check it

ninjab3s, 2024-04-08:
TX must be the address you want to send to, RX must be the address you receive on/bind to.