srsran/srsRAN_4G #1317: srsRAN 4G is not able to completely install after Installation of UHD 3.15 for USRP X310 and failed at 85% after execution of make command on linux ubuntu

HaiderHaideri, 2024-04-07:
**### Issue Description**

I have successfully installed RF front end drivers such as UHD 3.15 for detection of USRP X310 on virtual machine having OS Ubuntu and after it, I want to install open source srsRAN 4G software suite but unfortunately it did not go well and failed to install it completely. I have tried several times but every time it failed exactly at 85% during execution of make command. Till cmake ../ the process went sucessfull but after running make command, this error occured on both VM's. I am attaching snapshot here for reference.

### **## Setup Details ##**
OS : Ubuntu 22.04
RF Front end drivers : UHD 3.15
USRP : X310

### **## Expected Behavior ##**
When I am running make command in order to download and build srsRAN_4G, the below mentioned behavior is expected. 
[100%] Built target pyuhd_library 
[image]

### **## Actual Behaviour ##**
After make command, we need to install srsRAN_4G but this installation solely depends on successful execution of make command which ends at 100% but in my case, it failed at 85% with the below mentioned message. 
make[1]:  *** [Cmakefiles/Makefile2:12787:  srsenb/src/stack/mac/CmakeFiles/srsenb_mac.dir/all]  Error 2  make:  *** [Makefile:163:  all]   Error 2
[image]

### **## Steps to reproduce the problem ##**
git clone https://github.com/srsRAN/srsRAN_4G.git
cd srsRAN_4G
mkdir build
cd build
cmake ../
make

During make command, this error occurred when it comes to 85% 

### **## Additional Information ##**
I need to mention that on another VM, I have tried to install srsRAN 4G on another VM where I have not installed UHD 3.15 because no USRP is connected but surprisingly srsRAN 4G installed successfully there.

Alirezasf007, 2024-04-08:
Hi, 

> **### Issue Description**
> 
> I have successfully installed RF front end drivers such as UHD 3.15 for detection of USRP X310 on virtual machine having OS Ubuntu and after it, I want to install open source srsRAN 4G software suite but unfortunately it did not go well and failed to install it completely. I have tried several times but every time it failed exactly at 85% during execution of make command. Till cmake ../ the process went sucessfull but after running make command, this error occured on both VM's. I am attaching snapshot here for reference.
> 
> ### **## Setup Details ##**
> OS : Ubuntu 22.04 RF Front end drivers : UHD 3.15 USRP : X310
> 
> ### **## Expected Behavior ##**
> When I am running make command in order to download and build srsRAN_4G, the below mentioned behavior is expected. [100%] Built target pyuhd_library ![expected behaviour](https://private-user-images.githubusercontent.com/166223854/320276327-1f0ab74c-3ef0-4391-8972-89a75171dbec.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1NzgyNTIsIm5iZiI6MTcxMjU3Nzk1MiwicGF0aCI6Ii8xNjYyMjM4NTQvMzIwMjc2MzI3LTFmMGFiNzRjLTNlZjAtNDM5MS04OTcyLTg5YTc1MTcxZGJlYy5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOFQxMjA1NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZWM0MTU3N2ZlMTViNDA5MDQ5YmM5NjQ4ZjIzMzgzYTBkN2ZmODU3OGJiM2E2OGFhNGZkYjdmYWQ2NjQ5OThjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iTCvEFv3CZZdpbqjKeM0TGl78a1R35nVLGWqXDivXgE)
> 
> ### **## Actual Behaviour ##**
> After make command, we need to install srsRAN_4G but this installation solely depends on successful execution of make command which ends at 100% but in my case, it failed at 85% with the below mentioned message. make[1]: *** [Cmakefiles/Makefile2:12787: srsenb/src/stack/mac/CmakeFiles/srsenb_mac.dir/all] Error 2 make: *** [Makefile:163: all] Error 2 ![make command](https://private-user-images.githubusercontent.com/166223854/320276208-93033243-8594-47f9-bcc4-14e035b4c00f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1NzgyNTIsIm5iZiI6MTcxMjU3Nzk1MiwicGF0aCI6Ii8xNjYyMjM4NTQvMzIwMjc2MjA4LTkzMDMzMjQzLTg1OTQtNDdmOS1iY2M0LTE0ZTAzNWI0YzAwZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOFQxMjA1NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01N2IzMjYwOTE0ZDg2ZDYwNzJiZjQzMGNhODVlODA5MmVkNDMzNjczZjkyZTk1ZmUwYWQ0OGE1NmMwZjZkZjZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5Bi7p7UpU7H8x5igVAHbXpQpaALNhd8Tu2-95jYUX8o)
> 
> ### **## Steps to reproduce the problem ##**
> git clone https://github.com/srsRAN/srsRAN_4G.git cd srsRAN_4G mkdir build cd build cmake ../ make
> 
> During make command, this error occurred when it comes to 85%
> 
> ### **## Additional Information ##**
> I need to mention that on another VM, I have tried to install srsRAN 4G on another VM where I have not installed UHD 3.15 because no USRP is connected but surprisingly srsRAN 4G installed successfully there.

Hi, how much your vm's resource? Specifically the number of core and ram?

HaiderHaideri, 2024-04-08:
> Hi,
>
>
> Hi, how much your vm's resource? Specifically the number of core and ram?

its 8GB RAM and 8 core, however, issue resolved after increasing RAM.