srsran/srsRAN_4G #1334: TX while waiting for EOB, timed out...

LordyBass, 2024-05-14:
## Issue Description ##
When run 
$ sudo srsenb
TX while waiting for EOB, timed out...

## Setup Details ##
Im using USRP B210 with 22.04 Ubuntu in Oracle VM Virtual Box. Im also using Open5gs as core network and srsran as radio access network

## Expected Behavior ##
srsenb running succesfully

## Actual Behaviour ##
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit 254cc719a on branch HEAD.

Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-0ubuntu1~jammy1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1865.0 Mhz, UL=1770.0 MHz for cc_idx=0 nof_prb=50
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 2.09916 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 10.1058 >= 10.1057. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 14.0572 >= 13.0968. Starting new burst...

## Additional Information ##
Here is my configuration

## mme.yaml ##
logger:
  file:
    path: /var/log/open5gs/mme.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

mme:
  freeDiameter: /etc/freeDiameter/mme.conf
  s1ap:
    server:
      - address: 127.0.0.6
  gtpc:
    server:
      - address: 127.0.0.2
    client:
      sgwc:
- address: 127.0.0.3
      smf:
        - address: 127.0.0.4
  metrics:
    server:
      - address: 127.0.0.2
        port: 9090
  gummei:
    - plmn_id:
        mcc: 001
        mnc: 01
      mme_gid: 2
      mme_code: 1
  tai:
    - plmn_id:
        mcc: 001
        mnc: 01
      tac: 1
  security:
    integrity_order : [ EIA2, EIA1, EIA0 ]
   ciphering_order : [ EEA0, EEA1, EEA2 ]

## sgwu.yaml ##
logger:
  file:
    path: /var/log/open5gs/sgwu.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

sgwu:
  pfcp:
    server:
      - address: 127.0.0.6
    client:
#      sgwc:    # SGW-U PFCP Client try to associate SGW-C PFCP Server
#        - address: 127.0.0.3
  gtpu:
    server:
      - address: 127.0.0.6

## enb.conf ##
[enb]
enb_id = 0x19B
mcc = 001
mnc = 01
mme_addr = 127.0.0.6
gtp_bind_addr = 127.0.1.1
s1c_bind_addr = 127.0.1.1
s1c_bind_port = 0
n_prb = 50

## rr.conf ##
cell_list =
(
  {
    // rf_port = 0;
    cell_id = 0x01;
    tac = 1;
    pci = 1;
    // root_seq_idx = 204;
    dl_earfcn = 1800;
    //ul_earfcn = 21400;
    ho_active = false;
    //meas_gap_period = 0; // 0 (inactive), 40 or 80
    //meas_gap_offset_subframe = [6, 12, 18, 24, 30];
    // target_pusch_sinr = -1;
    // target_pucch_sinr = -1;
    // enable_phr_handling = false;
    // min_phr_thres = 0;
    // allowed_meas_bw = 6;
    // t304 = 2000; // in msec. possible values: 50, 100, 150, 200, 500, 1000, 2000
    // CA cells
    scell_list = (
      // {cell_id = 0x02; cross_carrier_scheduling = false; scheduling_cell_id = 0x02; ul_allowed >
    )

    // Cells available for handover
    meas_cell_list =
    (
      {
        eci = 0x19C02;
        dl_earfcn = 1800;
        pci = 2;
        //direct_forward_path_available = false;
        //allowed_meas_bw = 6;
        //cell_individual_offset = 0;
      }
    );

egg988, 2024-06-20:
Hi! I met the same problem "Tx while waiting for EOB, timed out...". I am using USRP X410. Have you fixed the problem? Thanks.

LordyBass, 2024-07-12:
Hello @egg988 !

For now, I am resolving the issue by using Ubuntu 22.04 installed natively, as I previously encountered the problem when using a virtual machine. I believe the issue lies with the virtual machine, which cannot provide optimal processor performance, resulting in errors. At this time, I have yet to find a solution for using the virtual machine to run srsenb optimally. I hope my this can help..

closed 2024-07-12 by LordyBass

egg988, 2024-08-12:
Hi @LordyBass !
Thanks a lot! I am using Ubuntu 20.04 and meet this problem. Could you please share the solution for using the virtual machine to run srsenb optimally? Maybe it helps. Thank you!

egg988, 2024-08-12:
Hi @LordyBass !
Thanks a lot! I am using Ubuntu 20.04. I agree that the issue results from inadequate computation power. I solved it by using only 25 PRB. Could you please share the solution for using the virtual machine to run srsenb optimally? Thank you!

Tonyhauatust, 2024-08-31:
I am encountering the same issue with X310 UHD driver version 4.5. Currently using n_prb 50 and default 1Gbps port. Have you guys tried raising the niceness etc as documented here https://docs.docker.com/engine/containers/resource_constraints/?

Tonyhauatust, 2024-08-31:
> Side topic: Encountering the same thing but with Docker.
> 
> I am encountering the same issue with X310 UHD driver version 4.5 OS version Ubuntu 20.02. Currently using n_prb 50 and default 1Gbps port. Have you guys tried raising the niceness etc as documented here https://docs.docker.com/engine/containers/resource_constraints/?

Follow up: 
By increasing the niceness to -20 the frequency of occurence is reduced. I think it is PC dependent too as one of my newly purchased PC gives none of these messages when running the same image. The newly purchased PC has not much processes running on it except the dockers etc.

lfarizav, 2024-10-26:
I have the same problem. I installed srsran on lxd and it does not work for the WARNING: Could not verify cpu0 scaling governor reason