srsran/srsRAN_4G #1339: Issues compiling srsRAN and srsGUI on Ubuntu 24.04

kschepps, 2024-05-28:
## Issue Description ##
I'm trying to install srsGUI and srsRAN_4G on Ubuntu 24.04. This is not a show-stopper as I can downgrade back to Ubuntu 22.04 which installs flawlessly every time on a fresh install, but I'm testing on newer PC hardware which does not seem to run very well on the older Linux firmware. I can't use srsRAN_Project as my setup uses the BladeRF which is not supported yet according to my research.

Does anyone have a fix to get srsRAN_4G to compile for Ubuntu 24.04?

## Setup Details ##

```
sudo apt install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev libboost-system-dev libboost-test-dev libboost-thread-dev libqwt-qt5-dev qtbase5-dev libusb-1.0-0-dev libusb-1.0-0 build-essential cmake libncurses5-dev libtecla1 libtecla-dev pkg-config git wget linux-tools-common linux-tools-generic terminator curl gnupg ca-certificates
var=$(uname -r) && eval "sudo apt install linux-tools-$var -y"
sudo cpupower frequency-set --governor performance

git clone https://github.com/Nuand/bladeRF.git
git clone https://github.com/srsRAN/srsGUI.git
git clone https://github.com/srsRAN/srsRAN.git

```

I also install mongodb, open5gs, and nodejs, but that should not be relevant to this issue.

bladeRF source compiles with no issues.

srsGUI:
```
cd srsGUI
mkdir build && cd build
cmake ../
make 
```

srsGUI error:
```
[ 90%] Building CXX object test/cxx/CMakeFiles/waterfallplot_test.dir/waterfallplot_test.cpp.o
In file included from /home/searf/Documents/srsGUI/test/cxx/waterfallplot_test.cpp:7:
/usr/include/boost/progress.hpp:23:3: error: #error This header is deprecated and will be removed. (You can define BOOST_TIMER_ENABLE_DEPRECATED to suppress this error.)
   23 | # error This header is deprecated and will be removed. (You can define BOOST_TIMER_ENABLE_DEPRECATED to suppress this error.)
      |   ^~~~~
In file included from /usr/include/boost/progress.hpp:29:
/usr/include/boost/timer.hpp:21:3: error: #error This header is deprecated and will be removed. (You can define BOOST_TIMER_ENABLE_DEPRECATED to suppress this error.)
   21 | # error This header is deprecated and will be removed. (You can define BOOST_TIMER_ENABLE_DEPRECATED to suppress this error.)
      |   ^~~~~
In file included from /usr/include/boost/smart_ptr/detail/requires_cxx11.hpp:9,
                 from /usr/include/boost/smart_ptr/scoped_ptr.hpp:13,
                 from /usr/include/boost/scoped_ptr.hpp:13,
                 from /home/searf/Documents/srsGUI/test/cxx/waterfallplot_test.cpp:5:
/usr/include/boost/progress.hpp:27:1: note: ‘#pragma message: This header is deprecated. Use the facilities in <boost/timer/timer.hpp> or <boost/timer/progress_display.hpp> instead.’
   27 | BOOST_HEADER_DEPRECATED( "the facilities in <boost/timer/timer.hpp> or <boost/timer/progress_display.hpp>" )
      | ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/timer.hpp:25:1: note: ‘#pragma message: This header is deprecated. Use the facilities in <boost/timer/timer.hpp> instead.’
   25 | BOOST_HEADER_DEPRECATED( "the facilities in <boost/timer/timer.hpp>" )
      | ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [test/cxx/CMakeFiles/waterfallplot_test.dir/build.make:76: test/cxx/CMakeFiles/waterfallplot_test.dir/waterfallplot_test.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:335: test/cxx/CMakeFiles/waterfallplot_test.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

srsGUI is not required for srsRAN so I also attempted to compile srsRAN without srsGUI

srsRAN:
```
cd srsRAN
mkdir build && cd build
cmake ../
make
```

srsRAN error:
[... 19 lines cut ...]
    inlined from ‘_OI std::copy(_II, _II, _OI) [with _II = asn1::rrc::meas_id_to_add_mod_s*; _OI = asn1::rrc::meas_id_to_add_mod_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:633:7,
    inlined from ‘void asn1::dyn_array<T>::resize(uint32_t, uint32_t) [with T = asn1::rrc::meas_id_to_add_mod_s]’ at /home/searf/Documents/srsRAN/lib/include/srsran/asn1/asn1_utils.h:248:18,
    inlined from ‘int test_meas_config()’ at /home/searf/Documents/srsRAN/lib/test/asn1/srsran_asn1_rrc_meas_test.cc:183:42:
/usr/include/c++/13/bits/stl_algobase.h:437:30: error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ forming offset 3 is out of the bounds [0, 3] [-Werror=array-bounds=]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In static member function ‘static _Up* std::__copy_move<_IsMove, true, std::random_access_iterator_tag>::__copy_m(_Tp*, _Tp*, _Up*) [with _Tp = asn1::rrc::quant_cfg_nr_r15_s; _Up = asn1::rrc::quant_cfg_nr_r15_s; bool _IsMove = false]’,
    inlined from ‘_OI std::__copy_move_a2(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:506:30,
    inlined from ‘_OI std::__copy_move_a1(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:533:42,
    inlined from ‘_OI std::__copy_move_a(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:540:31,
    inlined from ‘_OI std::copy(_II, _II, _OI) [with _II = asn1::rrc::quant_cfg_nr_r15_s*; _OI = asn1::rrc::quant_cfg_nr_r15_s*]’ at /usr/include/c++/13/bits/stl_algobase.h:633:7,
    inlined from ‘void asn1::dyn_array<T>::resize(uint32_t, uint32_t) [with T = asn1::rrc::quant_cfg_nr_r15_s]’ at /home/searf/Documents/srsRAN/lib/include/srsran/asn1/asn1_utils.h:248:18,
    inlined from ‘int test_meas_config()’ at /home/searf/Documents/srsRAN/lib/test/asn1/srsran_asn1_rrc_meas_test.cc:194:57:
/usr/include/c++/13/bits/stl_algobase.h:437:30: error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ forming offset 36 is out of the bounds [0, 36] [-Werror=array-bounds=]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [lib/test/asn1/CMakeFiles/srsran_asn1_rrc_meas_test.dir/build.make:76: lib/test/asn1/CMakeFiles/srsran_asn1_rrc_meas_test.dir/srsran_asn1_rrc_meas_test.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:8469: lib/test/asn1/CMakeFiles/srsran_asn1_rrc_meas_test.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

```

## Expected Behavior ##
This process works flawlessly every time on Ubuntu 22.04.4 using a fresh install

kschepps, 2024-05-31:
Thank you, @Lorenzoval! That worked!!

Here's the full set of commands for anyone else trying to compile srsRAN on Ubuntu 24.04. @Lorenzoval's fix should be added before the original cmake command. I also had to add ../ to one of the new cmake commands, but not the other.

```
sudo apt install gcc-11 g++-11

...

cd srsRAN
mkdir build && cd build

export CC=$(which gcc-11)
export CXX=$(which g++-11)
cmake ../ -B build
cmake --build build -j$(nproc)

cmake ../
make
make test
sudo make install
sudo ldconfig
srsran_install_configs.sh user
```

I still couldn't get srsGUI to work, but I'm probably going to phase that out on my setup anyways. It's way to resource intensive for just a representation that things are working. 

I'm going to use this instead:
```
echo 't' | sudo srsenb
```

blank-Planet, 2024-12-20:
Same problem here. srsRAN 4G does not compile in fresh installation of Ubuntu 24.04.01. I will follow [Lorenzoval](https://github.com/Lorenzoval) solution.

ltoenning, 2025-05-05:
I just encountered the same on Ubuntu 22.04. Instead of switching the compiler, I disabled the ``Werror``s (not tested with srsGUI). Not ideal, but seems to work.

``cmake -D CMAKE_CXX_FLAGS="-Wno-array-bounds -Wno-stringop-overflow -Wno-stringop-overread"``

sebinmpn, 2025-09-15:
what is the prifered os for srs ran