srsran/srsRAN_4G #1342: UE cannot find the LTE Network

fiveheadShinchan, 2024-05-29:
## Issue Description ##

I have followed the directions from the website: https://lists.srsran.com/mailman/listinfo/srsran-users
I have tried changing the EARFCN value to 3350, nof_prb to 25 and 50, and I have also checked the configuration files. I have also read similar tickets and the troubleshoot as well. 

## Setup Details ##

I am using an Oracle VM Virtual box running with DragonOS Focal (20.04) are out-of-the-box Lubuntu based x86_64 operating systems. I have allocated plenty of storage and 8 processors. 

I am using an Ettus 2450 antennas, USRP B210 with the GPSDO activated with external power the LED light is stable, I am using an open cell programmable SIM card with Pixel 3 XR phone. 

sudo srsepc
[image]

sudo srsenb
[image]

[image]

[image]

epc configuration file
[image]

enb configuration file
[image]

user_db.csv file 
[image]

## Expected Behavior ##
UE is able to find the network when looking for network operators

## Actual Behavior ##
UE is not able to see the network that was set up.

andrepuschmann, 2024-05-29:
It's not straight-forward run timing critical workloads out of a VM. The lates you see are likely causing the UE to not see the cell and thus it can't connect. I suggest you install bare-metal instead.

closed 2024-05-29 by andrepuschmann

fiveheadShinchan, 2024-05-31:
Hello,

Thank you for the quick response. I was wondering if it was not possible to
run on vm. I actually ended up using the wrong USRP B210, without the GPSDO
attached. When I attached the one with GPSDO, I only  got lates of 5-7. The
UE is still not able to find the network.