srsran/srsRAN_4G #1346: srsRAN_4G fails to build

pkeyela, 2024-06-04:
# Issue Description
I'm building the srsRAN-4G from source but it repeatedly throws this error. I'm using a VM with Ubuntu 24 as OS, 4 CPU cores and 8GB as RAM. I thought increasing the ressources of the VM would help, but it didn't. I would be very grateful for any help out. On the same machine I built and installed srsRAN_Project whithout any issue.
[image]

icecool1987, 2024-06-08:
Thank this might be of help [issue 1339](https://github.com/srsran/srsRAN_4G/issues/1339)

rob-the-dude, 2024-06-17:
FYI, I found indications online that these were spurious warnings due to compiler bugs in GCC.  I made a handful of modifications to various CMakeLists.txt files like this:

`set_source_files_properties(FOO.cc PROPERTIES COMPILE_FLAGS "-Wno-array-bounds")
`
or 

`set_source_files_properties(FOO.cc PROPERTIES COMPILE_FLAGS "-Wno-array-bounds -Wno-stringop-overflow")
`
or even

`set_source_files_properties(FOO.cc PROPERTIES COMPILE_FLAGS "-Wno-array-bounds -Wno-stringop-overflow -Wno-stringop-overread")
`

And I was able to get it to build (and run).  I can provide a full 'git diff' if that helps.  But I think the real fix requires an updated GCC (I think the bug is still open there).