srsran/srsRAN_4G #1360: the spectrum bandwidth of the Transmitted 4G signal in embms mode is significantly larger

nenad38111, 2024-07-11:
## Issue Description ##
The spectrum is definitely wider in this mode comparing to a 'non-embms' mode.
For PRB=15 (3MHz), the spectrum in embms mode is about 1-1.5MHz wider.

Have this setup and is basically working, but the shape of the spectrum 'outside the desired bandwidth' (in my case 3MHz) is not what I expected, especially as it looks normal in non-embms mode.

## Setup Details ##
srsmbms
srsepc
srsenb                                   srsue
B210 (TX) ------------------- B210 (RX)
UHD driver 3.15LTS
ubuntu 22.04
source code compiled for srsran_4G

All the config files matching those outlined here:
https://docs.srsran.com/projects/4g/en/latest/app_notes/source/embms/source/index.html

## Expected Behavior ##
Spectrum to be the same bandwidth in non embms as in the embms mode.

## Actual Behaviour ##
Spectrum in embms mode has significantly higher side lobes causing the BW of the spectrum to be much wider.

## Steps to reproduce the problem ##
Just follow the instructions on the 
https://docs.srsran.com/projects/4g/en/latest/app_notes/source/embms/source/index.html

## Additional Information ##
Have tried with B210, but also BladeRF. It is about the same.

yagoda, 2024-07-11:
Hi,

The reference signals are significantly different in eMBMS, they span the entire band whereas the cell reference signals in unicast are more sparse, try full traffic in both modes and compare the spectrum mask in that scenario.

Regards,
Justin.

nenad38111, 2024-07-11:
Not sure if this picture is going to show up, but in non embms mode, the spectrum is well contained in 3MHz

[image]

yellow trace - embms, blue trace embms (max hold)
green trace - non embms, red trace non embms (max hold).

yagoda, 2024-07-12:
Can you please share the logs & terminal outpu of the enb for both embms and non-embms mode?

nenad38111, 2024-07-12:
Yes of course....and thank you for looking into this.

[enb_embms_enabled.log](https://github.com/user-attachments/files/16195529/enb_embms_enabled.log)
[enb_no_embms.log](https://github.com/user-attachments/files/16195530/enb_no_embms.log)
[terminal_log_embms_enabled.txt](https://github.com/user-attachments/files/16195535/terminal_log_embms_enabled.txt)
[terminal_log_no_embms.txt](https://github.com/user-attachments/files/16195536/terminal_log_no_embms.txt)

nenad38111, 2024-07-18:
One more picture.... I have tried everything I could think off, but it is always the same thing...in embms mode the spectrum is significantly wider than it should be.

[image]

nenad38111, 2024-07-23:
I don't know if related, but eMBMS (for 4G) is not working for PRB=6. It does work for other PRBs (15,25,50).