srsran/srsRAN_4G #1367: I have two mobile with programmable Simcard, Registered on SRSRAN enb, How can Call or SMS between two mobile?

stackprogramer, 2024-07-28:
I have two mobile with programmable Simcard, Registered on SRSRAN enb, How can Call or SMS between two mobile?
I have two mobile with Programmable Simcard, They can show SRSRAN network, They are registered on the network, my question is which is the phone number for ue1 ue2 that we have set to file user_db file in SRSRAN. I want to make a call or make sms between two mobile. VolLTE is supported on SRSRAN project?
Thanks in advance
#                                                                                           
# .csv to store UE's information in HSS                                                     
# Kept in the following format: "Name,Auth,IMSI,Key,OP_Type,OP/OPc,AMF,SQN,QCI,IP_alloc"  
#                                                                                           
# Name:     Human readable name to help distinguish UE's. Ignored by the HSS                
# Auth:     Authentication algorithm used by the UE. Valid algorithms are XOR               
#           (xor) and MILENAGE (mil)                                                        
# IMSI:     UE's IMSI value                                                                 
# Key:      UE's key, where other keys are derived from. Stored in hexadecimal              
# OP_Type:  Operator's code type, either OP or OPc                                          
# OP/OPc:   Operator Code/Cyphered Operator Code, stored in hexadecimal                     
# AMF:      Authentication management field, stored in hexadecimal                          
# SQN:      UE's Sequence number for freshness of the authentication                        
# QCI:      QoS Class Identifier for the UE's default bearer.                               
# IP_alloc: IP allocation stratagy for the SPGW.                                            
#           With 'dynamic' the SPGW will automatically allocate IPs                         
#           With a valid IPv4 (e.g. '172.16.0.2') the UE will have a statically assigned IP.
#                                                                                           
# Note: Lines starting by '#' are ignored and will be overwritten                           
ue2,mil,001011111111111,12345678901234567890123456789012,opc,212e3b94279cb0f8095a55e8ef5569f7,9000,00000000630a,9,dynamic

nishant412, 2024-09-23:
srsRAN doesn't support the IMS bearer, and therefore only allows connection to data PDN. Therefore you can't do VoLTE. However there is an alternate implementation that uses srsenb with Open5gs epc core and kamilio IMS SIP server. That would do the job for you.

herlesupreeth, 2024-09-24:
Correction to what @nishant412 said:

In order to achieve 4G calling functionality you need IMS which is a separate network entity and is not part of the EPC. srsEPC does not have support for signalling towards IMS in order to support IMS calling/registration. However, with the help of open5gs EPC + kamailio IMS you can achieve 4G calling using srsRAN eNB since srsRAN eNB supports setting up of bearers required for 4G calling.