srsran/srsRAN_4G #1375: UE suddenly release and go to Idle mode

faratell, 2024-08-18:
Hello. 
I have installed srsRAN_4G in two machines In one machine I run srsepc (192.168.75.185) and in others I run srsenb and srsepc.
eNB connection to EPC is ok. srsue is also connected to eNB, but after some seconds it releases and went to idle mode.

%%%%%%
sudo srsepc /root/.config/srsran/epc.conf

Built in Release mode using commit fa56836b1 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /root/.config/srsran/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf999, MNC: 0xff55
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:999, MNC:55
S1 Setup Request - TAC 1, B-PLMN 0x99f955
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x45d54d88
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 1
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: false
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 999550000000002
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 999550000000002
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 999550000000002
Sending ESM information request
UL NAS: Received ESM Information Response
ESM Info: APN internet
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 999550000000002
Creating Session Response -- MME control TEID: 1
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1
SPGW: Allocated User TEID 1
SPGW: Allocate UE IP 172.16.0.2
Received Create Session Response
Create Session Response -- SPGW control TEID 1
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.2 to IMSI 999550000000002
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x1; eNB GTP-U Address 192.168.75.149
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 999550000000002
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Sending EMM Information
Received UE Context Release Request. MME-UE S1AP Id 1
There are active E-RABs, send release access bearers request
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Received UE Context Release Complete. MME-UE S1AP Id 1
UE Context Release Completed.

%%%%
sudo ./srsue ue_zmq.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue_zmq.conf...

Built in Release mode using commit ec29b0c on branch master.

srsLog error - Unable to create log file "/tmp/ue.log": Permission denied
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=99955, TAC=1
Random Access Transmission: seq=41, tti=341, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Network attach successful. IP: 172.16.0.2
Software Radio Systems RAN (srsRAN) 18/8/2024 8:0:51 TZ:0
Received RRC Connection Release (releaseCause: other)
RRC IDLE

%%%%
udo ./srsenb enb.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in Release mode using commit ec29b0c on branch master.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
RACH:  tti=341, cc=0, pci=1, preamble=41, offset=0, temp_crnti=0x46
User 0x46 connected
t
Enter t to stop trace.

               -----------------DL----------------|-------------------------UL-------------------------
rat  pci rnti  cqi  ri  mcs  brate   ok  nok  (%) | pusch  pucch  phr  mcs  brate   ok  nok  (%)    bsr
lte    1   46   15   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   46   15   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
Disconnecting rnti=0x46.

faratell, 2024-08-18:
I have also test it with open5gs core and same result. After some seconds UE release. 

%%%%%
mme log:
08/18 11:30:29.758: [mme] INFO: eNB-S1 accepted[192.168.75.185]:39269 in s1_path module (../src/mme/s1ap-sctp.c:114)
08/18 11:30:29.758: [mme] INFO: eNB-S1 accepted[192.168.75.185] in master_sm module (../src/mme/mme-sm.c:108)
08/18 11:30:29.758: [mme] INFO: [Added] Number of eNBs is now 1 (../src/mme/mme-context.c:2557)
08/18 11:30:29.758: [mme] INFO: eNB-S1[192.168.75.185] max_num_of_ostreams : 30 (../src/mme/mme-sm.c:150)
08/18 11:30:36.286: [mme] INFO: eNB-S1[192.168.75.185] connection refused!!! (../src/mme/mme-sm.c:166)
08/18 11:30:36.286: [mme] INFO: [Removed] Number of eNBs is now 0 (../src/mme/mme-context.c:2591)
08/18 11:47:26.901: [mme] INFO: eNB-S1 accepted[192.168.75.149]:35316 in s1_path module (../src/mme/s1ap-sctp.c:114)
08/18 11:47:26.901: [mme] INFO: eNB-S1 accepted[192.168.75.149] in master_sm module (../src/mme/mme-sm.c:108)
08/18 11:47:26.901: [mme] INFO: [Added] Number of eNBs is now 1 (../src/mme/mme-context.c:2557)
08/18 11:47:26.901: [mme] INFO: eNB-S1[192.168.75.149] max_num_of_ostreams : 30 (../src/mme/mme-sm.c:150)
08/18 11:47:38.994: [mme] INFO: InitialUEMessage (../src/mme/s1ap-handler.c:402)
08/18 11:47:38.994: [mme] INFO: [Added] Number of eNB-UEs is now 1 (../src/mme/mme-context.c:4433)
08/18 11:47:38.994: [mme] INFO: Unknown UE by S_TMSI[G:2,C:26,M_TMSI:0x4c872aa1] (../src/mme/s1ap-handler.c:482)
08/18 11:47:38.995: [mme] INFO:     ENB_UE_S1AP_ID[1] MME_UE_S1AP_ID[2] TAC[1] CellID[0x19b01] (../src/mme/s1ap-handler.c:578)
08/18 11:47:38.995: [mme] INFO: Unknown UE by GUTI[G:1,C:26,M_TMSI:0x4c872aa1] (../src/mme/mme-context.c:3287)
08/18 11:47:38.995: [mme] INFO: [Added] Number of MME-UEs is now 1 (../src/mme/mme-context.c:3089)
08/18 11:47:38.995: [emm] INFO: [] Attach request (../src/mme/emm-sm.c:412)
08/18 11:47:38.995: [emm] INFO:     GUTI[G:1,C:26,M_TMSI:0x4c872aa1] IMSI[Unknown IMSI] (../src/mme/emm-handler.c:245)
08/18 11:47:39.018: [emm] INFO: Identity response (../src/mme/emm-sm.c:382)
08/18 11:47:39.018: [emm] INFO:     IMSI[999550000000002] (../src/mme/emm-handler.c:409)
08/18 11:47:39.088: [mme] INFO: [Added] Number of MME-Sessions is now 1 (../src/mme/mme-context.c:4447)
08/18 11:47:39.400: [emm] INFO: [999550000000002] Attach complete (../src/mme/emm-sm.c:1298)
08/18 11:47:39.400: [emm] INFO:     IMSI[999550000000002] (../src/mme/emm-handler.c:283)
08/18 11:47:39.400: [emm] INFO:     UTC [2024-08-18T08:17:39] Timezone[0]/DST[0] (../src/mme/emm-handler.c:290)
08/18 11:47:39.400: [emm] INFO:     LOCAL [2024-08-18T11:47:39] Timezone[12600]/DST[0] (../src/mme/emm-handler.c:294)
08/18 11:48:13.264: [mme] INFO: UE Context Release [Action:2] (../src/mme/s1ap-handler.c:1819)
08/18 11:48:13.264: [mme] INFO:     ENB_UE_S1AP_ID[1] MME_UE_S1AP_ID[2] (../src/mme/s1ap-handler.c:1820)
08/18 11:48:13.264: [mme] INFO:     IMSI[999550000000002] (../src/mme/s1ap-handler.c:1823)
08/18 11:48:13.264: [mme] INFO: [Removed] Number of eNB-UEs is now 0 (../src/mme/mme-context.c:4440)

%%%% srsue
sudo ./srsue ue_zmq.conf
[sudo] password for faraz: 
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue_zmq.conf...
srsLog error - Unable to create log file "/tmp/ue.log": Permission denied

Built in Release mode using commit ec29b0c on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=99955, TAC=1
Random Access Transmission: seq=21, tti=341, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Network attach successful. IP: 10.45.0.7
 nTp) 18/8/2024 8:17:39 TZ:65
Received RRC Connection Release (releaseCause: other)
RRC IDLE

%%%

faratell, 2024-08-18:
I also checked with physically simcard. it attached successfully but after a while it release

frankist, 2024-08-18:
Have you checked that it is not just the UE inactivity timer? The eNB will release the UE automatically if it has no traffic.

faratell, 2024-08-19:
thanks . my problem is solved by defining rrc_inactivity_timer = 3000000

closed 2024-08-19 by herlesupreeth