srsran/srsRAN_4G #1380: Error predecoding multiplex: not implemented for 4 Tx ports, Not Implemented (nof_tb=1, pinfo=7)

Pramodsmvdu, 2024-09-06:
Hello everyone,
I hope everyone is doing well .
Need some expert advice here ->

While running the test with 10 MHZ, 20 MHZ  with some of the cell tower , I see below error.

**/srsRAN_4G/lib/src/phy/phch/ra_dl.c:517: Not Implemented (nof_tb=1, pinfo=7)
srsRAN_4G/lib/src/phy/phch/ra_dl.c:592: Configuring MIMO PMI
srsRAN_4G/lib/src/phy/phch/ra_dl.c:517: Not Implemented (nof_tb=1, pinfo=7)
/srsRAN_4G/lib/src/phy/phch/ra_dl.c:592: Configuring MIMO PMI
srsRAN_4G/lib/src/phy/mimo/precoding.c:1853: Error predecoding multiplex: not implemented for 4 Tx ports**

or sometime..  

srsran_4g/lib/src/phy/phch/ra_dl.c:519: Not Implemented (nof_tb=1, pinfo=5)
/srsran_4g/lib/src/phy/phch/ra_dl.c:594: Configuring MIMO PMI

After looking into code i see that..
**srsran_4g/lib/src/phy/phch/ra_dl.c** 

/* Translates Precoding Information (pinfo) to Precoding matrix Index (pmi) as 3GPP 36.212 Table 5.3.3.1.5-4 */
static int config_mimo_pmi(const srsran_cell_t* cell, const srsran_dci_dl_t* dci, srsran_pdsch_grant_t* grant)
{
  uint32_t nof_tb = grant->nof_tb;
  if (grant->tx_scheme == SRSRAN_TXSCHEME_SPATIALMUX) {
    if (nof_tb == 1) {
      if (dci->pinfo > 0 && dci->pinfo < 5) {
        **grant->pmi = dci->pinfo - 1;
      } else {
         **ERROR("Not Implemented (nof_tb=%d, pinfo=%d)", nof_tb, dci->pinfo);**
        return -1;
      }**
    } else {

Seem this is not implemented in srsran software for pinfo > 5.  currently i am using  TAG release_23_04 .

Any suggest if support for pinfo >5 has been added ?

or any workaround to avoid this situation to attach the UE ?  can i disable mimo completely  to let us attach ? 
any suggestion to try changes in ue capability etc .

Thank you in advance !

Pramodsmvdu, 2024-09-17:
Hello everyone,
Any input on this ? looking for your feedback on this problem.
Thank you !

tavako, 2025-07-16:
I have the same problem , I'm using a 1tx , 1rx setup ue , trying to connect to a 4 port enode b , Have you found any solution for this problem ? , should we implement this part or is it a problem from higher layers ?