srsran/srsRAN_4G #1384: Decode ue_mac.pcap with wireshark

Pramodsmvdu, 2024-09-13:
I followed the steps mentioned at https://docs.srsran.com/projects/4g/en/rfsoc/general/source/5_troubleshooting.html to enable to see the message in wireshark.  in am running srsue with lte protocol.
i tried.. 

Packet capture files (pcaps) can be viewed using Wireshark (www.wireshark.org). pcaps are encoded in compact MAC-LTE and MAC-NR form. They can be found in the /tmp folder where other logs are located. To view in wireshark, edit the preferences of the DLT_USER dissector.

To decode MAC pcaps add an entry with the following:

DLT=149

Payload Protocol=udp

Further, enable the heuristic dissection in UDP under: Analyze > Enabled Protocols > MAC-LTE > mac_lte_udp and MAC-NR > mac_nr_udp

Using the same filename for mac_filename and mac_nr_filename writes both MAC-LTE and MAC-NR to the same file allowing a better analysis.

To decode NAS pcaps add and entry with the following:

DLT=148

Payload Protocol=nas-eps

how to enable mac, nas decoding correctly ?

[image]

Mouradnetworking, 2024-09-16:
Hello, Go to Analyse > Enable Protocols and enable all protocols

Pramodsmvdu, 2024-09-16:
> Hello, Go to Analyse > Enable Protocols and enable all protocols

Thank you ! I tried enabling all protocol but still i see the same output as above.
Just to add , I am trying to open the pcap on window pc with wireshark 4.0.6.
on wireshark on ubuntu i don't see the option to configure   DLT=148 Payload Protocol=nas-eps .
anything specific to do ? 
i hope it should work on wireshark on window or ubuntu both ?

Mouradnetworking, 2024-09-16:
I'm not sure if you're saying that you couldn't find the DLT user configuration or that you're unable to add entries to the DLT user. Could you clarify?

Ps: it works also on ubuntu.

Pramodsmvdu, 2024-09-16:
Hlelo @Mouradnetworking  I am using  Wireshark 3.4.2  version.
I am following page to add config  https://docs.srsran.com/projects/4g/en/rfsoc/general/source/5_troubleshooting.html 
To view in wireshark, edit the preferences of the DLT_USER dissector.

i see this option in window wireshark under edit-> preference->DLT_USER

but i don't see this option as  edit-> preference->DLT_USER  on Wireshark 3.4.2  on ubuntu . 
so may be there is way to configure it in other way ?

Mouradnetworking, 2024-09-16:
No problem! Right-click on one of the packets, then go to Protocol Preferences > DLT User > Open DLT User Preferences.

Pramodsmvdu, 2024-09-16:
@Mouradnetworking  Thanks !  I am able to add the config. now :).  I am not seeing  the option to enable protocol : Analyze > Enabled Protocols > MAC-LTE > mac_lte_udp  . what option should i use.. i am unable to find enable protocol option ..

Mouradnetworking, 2024-09-16:
Go to Analyze > Enable protocols > Enable all.

Mouradnetworking, 2024-09-16:
Try to open ubuntu with sudo

Pramodsmvdu, 2024-09-16:
@Mouradnetworking Any recommended wireshark version for this to decode successfully ?

Mouradnetworking, 2024-09-16:
can try this again ? 

DLT = 147 --> mac-lte-framed 
DLT = 148 --> nas-eps 
DLT = 149 --> udp
DLT = 150 --> s1ap

Pramodsmvdu, 2024-09-16:
I tried. still the result are same.

Pramodsmvdu, 2024-09-16:
Thank you @Mouradnetworking   for your help with detailed input.
 after trying multiple things. was able to figure out the problem.
not sure why but with Wireshark 3.4.2  version same setting was not working.
I updated to Wireshark 4.4.0 and same setting worked without any problem.
[image]

Mouradnetworking, 2024-09-16:
Hey again, No problem GL.

Best Regards,

closed 2024-09-17 by Pramodsmvdu