srsran/srsRAN_4G #1389: srsRAN looks for wrong path for config files and LD_LIBRARY_PATH is not set by the make install

armin485, 2024-09-22:
## Issue Description ##
Dear srsRAN commity.
First of all I want to thank you for this wonderful project.
I've been installing srsRAN_4G from code with zmq as RF device and it's working great. However, since the applications (EPC, ENB and UE) required to be run with sudo, all of them are trying to read config files in root directory but the build and installation process have been done with my normal user (ubuntu). The typical error I got is:

`ubuntu@zbook:~$ sudo srsepc

Built in Release mode using commit ec29b0c1f on branch master.

---  Software Radio Systems EPC  ---

Couldn't open , trying /root/.config/srsran/epc.conf
Couldn't open /root/.config/srsran/epc.conf either, trying /etc/srsran/epc.conf
Failed to read EPC configuration file /etc/srsran/epc.conf - exiting`

After copying the config files from home directory (ubuntu) into root, it works.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
All nework components are running on same machine (my laptop) with zmq as radio transceiver. I've followed official installation and running instruction.

## Expected Behavior ##
sudo srsepc to be run without setting config directories

## Actual Behaviour ##
srsepc, srsenb and srsue all fail to find config files

## Steps to reproduce the problem ##
Installation and building from source including zmp. Follow the official instruction.

rcbarke, 2024-09-25:
I am running into this same issue with the sRsRAN version documented here. Oddly enough, it seems to be working fine on one of my node VMs, and is broken on my OAIC RIC VM. This is very strange as I followed exactly the same steps in both environments yet my entire /root/.config/srsran directory is missing on the RIC VM. I am going to blow it up and rebuild it, though deploying I'd greatly appreciate an explanation if someone has one. Thanks!
https://openaicellular.github.io/oaic/srsRAN_installation.html

rcbarke, 2024-09-25:
Actually, I spoke too soon. I have no clue why this happened, but the config files are present on my machine, they're just located within my home directory and not root directory.

To fix:
sudo -i
mkdir .config
cp -r ~/.config/* /root/.config/

Then verify that the below returns files (ex: enb.conf):
sudo ls -la /root/.config/srsran/

Weird bug.