srsran/srsRAN_4G #1404: Does nr_cell_search_rf work for band 28?

Carcalmoa, 2024-10-16:
## Issue Description ##
I would like to scan band 28 using nr_cell_search_rf.

## Expected Behavior ##
Get an output similar to cell_search:

[image]

## Actual Behaviour ##
[image]

## Steps to reproduce the problem ##
./nr_cell_search_rf --bands 28 --rf.device_name uhd --rf.device_args "type=b200, serial=xxxx" --rf.log_level info --log_level info --rf.rx_gain 50

## Additional Information ##
I don't understand why the central frequency is initialized to 0. I also tried to put it as a parameter and to use the value 780.5MHz as the downlink frequency range for band 28, which goes from 758 MHz to 803 MHz. I have checked that the SSB frequencies calculated for this band are correct following TS38.104 Table 5.4.3.1-1. Could someone help me? I would also be grateful if someone could suggest other ways of scanning this band using software. Thank you all!

[image]

LuxeZN, 2024-10-22:
I would try running with `sudo`, I would also try running the other tests. `./nr_cell_search_test` seems to work for me and get the result you want. 
You also are supposed to pass in a config file, but regardless I wasn't able to get that script to work on my end either.
```
sudo ./nr_cell_search_test --rf.device_name uhd --rf.device_args "type=b200" --srate 23040000 --ssb_scs 15 --meas_len_ms 1000 --meas_period_ms 1000 ~/.config/srsran/ue_20mhz_b210.conf
Active RF plugins: libsrsran_rf_uhd.so
Inactive RF plugins: 
Opening 1 channels in RF device=uhd with args=type=b200
Supported RF device list: UHD file
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.6.0.HEAD-0-g50fa3baa
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.

Warning: TX gain was not set. Using open-loop power control (not working properly)

-- Statistics:

    Probability of detection: 0.000000
  Probability of false alarm: 0.000000
2024-10-22T18:21:22.218853 [PHY    ] [W] [  999] NR intra frequency performance 0 Msps
```

nebukadnezar, 2025-01-14:
Looks like this is broken - badly. nr_cell_search_rf gives me the same error as the OPs for every band other than 78. And even for 78, it doesn't seem to scan the actual band but just goes up from 3.3 GHz until I kill it after passing 6 GHz... n78 should go from 3.3 - 3.8 GHz.