srsran/srsRAN_4G #1412: Issue with srsue getting stuck on 'Attaching UE'

adithya-anand-999, 2024-10-27:
## Issue Description ##
[Describe the issue in detail]
I am doing RAN slicing on the srsRAN_Project gNB with a Open5GS core network. I am able to get both the core and gNB to connect with the changes required for RAN slicing but I am having some issues with the srsue connecting with the gNB. For context I am using zmq, planning right now to have at least 1 srsue on each of the 2 sices on my gNB. I have also registered each srsue to the core network through the GUI for open5GS. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Installation of Open5GS, srsRAN_Project, and srsRAN_4G. Then following the below steps on modifying the config files. 
## Expected Behavior ##
[What you expect to happen]
srsue should connect to gNB. 

## Actual Behaviour ##
[What happens instead e.g. error message]
srsue gets stuck on 'Attaching UE' step. 
[image]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
All Config file modifications are as follows
Open5GS: amf.yaml
[image]

Open5GS: nssf.yaml
[image]
The comments at the end show another way I tried to make the slices, both approaches lead to the gNB connecting. 

srsRAN_Project: gNB config file
[image]

srsRAN_4G: srsue config file, srsue should be 5G enabled thus support network slicing from the research i have done. I have also tried not having slicing in general and commenting out that section, resulted in the same issue. 
[image]

open5GS GUI srsue registration
[image]
[image]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

LuxeZN, 2024-10-28:
What happens if you change the `srate` to 23.04 in srsUE?

adithya-anand-999, 2024-10-28:
yes I caught that right after I posted this, did not change anything. 

[image]

LuxeZN, 2024-10-28:
I don't think this will fix your problem but you are going to want the `apn` section to be "internet" instead of "srsapn".

adithya-anand-999, 2024-10-28:
as you said doesnt let the srsue connect, but thanks

LuxeZN, 2024-11-04:
What happens if you use these configuration files?
Configs:
gNB - [gnb_conf.txt](https://github.com/user-attachments/files/17618305/gnb_conf.txt)
UE - [ue_conf.txt](https://github.com/user-attachments/files/17618307/ue_conf.txt)

adithya-anand-999, 2024-11-04:
Ah I see, so this is for getting srsue's to connect over ZMQ. 

Thanks, this actually might help. As of now I have moved away from using the Open5GS repo and am now using the Open5GS docker container provided by srsRAN because even without slices my srsue's were not connecting. 

With the docker they worked immediately. 

Also have you done any work using multiple srsue's, currently I am using a broker (GNU Radio Companion) was wondering if there was any way for me to not have to use that.

LuxeZN, 2024-11-04:
I haven't done any work with multiple srsue's. I've seen people here get multiple UE's to work without a broker, and srsRAN has actually tested up to 64 UE's at once here https://github.com/srsran/srsRAN_Project/discussions/40. You should just be able to give the second srsue new configuration parameters (IMSI, etc.) and it should work? I haven't tried this though but I think that should work.

LuxeZN, 2024-11-04:
Have you tried just changing the parameters and connecting another UE? I would just try that and see if it will work, even if the documentation says otherwise, if not let me know and I can look further into the docs. Also I do a lot of work with USRPs so I'm not sure about a lot of the ZMQ stuff, especially with multiple UE's there.

adithya-anand-999, 2024-11-04:
I was in the process of trying till I ran into the issue that started this thread. 

The thread you shared actually does have some people talking about wanting to connect multiple srsue's over ZMQ and the consensus from the srsRAN team was to use the Broker setup with GNU radio as laid out in the specs. 

I do have a working system now, albeit with a Docker Core, so when I get the time I can experiment.

LuxeZN, 2024-11-04:
> I was in the process of trying till I ran into the issue that started this thread.
>
> The thread you shared actually does have some people talking about wanting to connect multiple srsue's over ZMQ [...]

Gotcha, I don't have any experience with Open5GS actually. The core network I have always used is free5gc, both containerized and not. Post results here please, would like to see what you can get/if you encounter any more problems.

adithya-anand-999, 2024-11-04:
sounds good, will keep you posted.

adithya-anand-999, 2024-11-07:
> Gotcha, I don't have any experience with Open5GS actually. The core network I have always used is free5gc, both containerized and not. Post results here please, would like to see what you can get/if [...]

Ended up solving the issue, right now system has 1 Docker Open5GS core, 1 srsRAN gNB with 2 slices, and 3 srsue's. 2 srsue's on 1 slice, and 1 on the other slice.

LuxeZN, 2024-11-07:
> Ended up solving the issue, right now system has 1 Docker Open5GS core, 1 srsRAN gNB with 2 slices, and 3 srsue's. 2 srsue's on 1 slice, and 1 on the other slice.

Awesome! Do you have the configs you used for the UE's and gNB?

adithya-anand-999, 2024-11-08:
Yes I do, they are all in the zip. 
[config files from RAN slicing with srsue.zip](https://github.com/user-attachments/files/17673022/config.files.from.RAN.slicing.with.srsue.zip)

closed 2024-11-09 by adithya-anand-999

LuxeZN, 2026-01-15:
> I got the same problem too,But i use USRP B210 instead of ZMQ. msg1,msg2 all ok,but the msg3 is wrong,as you said CRC all failed.And the ue is stuck with Random access.as the following pics. I've tried to change the tx/rx，and i checked if the parameter between gNB and UE is same too.But I have no clue whats wrong with my config,could you please help me check.Thanks!!!!
> 
> <img alt="Image" width="1317" height="607" src="https://private-user-images.githubusercontent.com/254214603/534935285-99c69e45-0842-4f66-b4c5-73d8d8b5a110.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njg0OTM5MDAsIm5iZiI6MTc2ODQ5MzYwMCwicGF0aCI6Ii8yNTQyMTQ2MDMvNTM0OTM1Mjg1LTk5YzY5ZTQ1LTA4NDItNGY2Ni1iNGM1LTczZDhkOGI1YTExMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDExNVQxNjEzMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNjFjN2NkMzkxZWE1YWE0MTQzMzY3N2M4ZTdkNTQzZTEwZjVkNzlkMjMxMDRlMjZiOGVjNzFmMzRhZWNkM2U3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Ez5qeAVmmxv4-TAniaZ5657qrT_sBCeNrVrTFaXaYIc"> <img alt="Image" width="1787" height="567" src="https://private-user-images.githubusercontent.com/254214603/534935339-0adb11ad-2762-4467-9992-aa8d4ebc37b8.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njg0OTM5MDAsIm5iZiI6MTc2ODQ5MzYwMCwicGF0aCI6Ii8yNTQyMTQ2MDMvNTM0OTM1MzM5LTBhZGIxMWFkLTI3NjItNDQ2Ny05OTkyLWFhOGQ0ZWJjMzdiOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDExNVQxNjEzMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMTlmNDJmMjgzMWY2ODAyOWIxNjUyZmEwZjA3YTE0NjAwNDg4ZWFkYWVkMjljMjFlMjdhMjQ4ODJmYjk0N2U0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.EclHouly0qy-RAcyvTkjP9DJguyPz-6t-iQokTIhlmQ">
> [ue_5g.txt](https://github.com/user-attachments/files/24580930/ue_5g.txt)
> 
> [final_gnb.txt](https://github.com/user-attachments/files/24580933/final_gnb.txt)

Have you tried running the srsRAN performance script like is suggested in the UE logs?