srsran/srsRAN_4G #1422: IQ files using ENB

erez1012, 2024-11-15:
I want to create IQ files using ENB and then transmit them using basic SDR, is ENB able to produce such files?

matth3wh, 2024-11-19:
I believe that you can use GNU Radio to dump a copy of the IQ samples into a file. Would that work for you? You would need to run the whole system with ZMQ (https://docs.srsran.com/projects/4g/en/next/app_notes/source/zeromq/source/), otherwise I do not think the eNB will start transmitting.

achkonsa, 2025-01-03:
i have a similar request. i want to decode MIB from an existing IQ file... have you succeed in your request?

erez1012, 2025-01-03:
No

windy-zzf, 2025-04-01:
> i have a similar request. i want to decode MIB from an existing IQ file... have you succeed in your request?

Did you succeed in doing so? Can you share that?

erez1012, 2025-04-01:
No, I did not.

-- 

בכבוד רב,

ארז דניאלי

Best Regards,

Erez Danieli

Electrical,RF Engineering

Particle Accelerator, LINAC

SCADA Security

Hardware Hacking

ISRAEL

Ariel University Center of Samaria
Tel:  +972-054-4711591
Tel:  +972-03-9066603
Fax: +972-054-4711591

fzybot, 2025-04-17:
What if you run with ZeroMQ and get IQ-samples in zmq_send?
```
/srsRAN_4G/lib/src/phy/rf/rf_zmq_imp_tx.c
n = zmq_send(q->sock, buf, (size_t)sample_sz * nsamples, 0);
```