srsran/srsRAN_4G #1429: Iperf uplink data throughput low

matth3wh, 2024-11-27:
## Issue Description ##
I have an srsRAN 4G lab setup with two USRP N310s. I am trying to verify that we get the expected uplink and downlink data rates. To test the rates, I am using iperf3 in the normal TCP mode. For the bandwidth I am starting with (25 PRBs which is 5 Mhz), I get the expected rate of about 19 Mbps. However, when I test the uplink, I get highly variable data rates, much lower than the expected 12 Mbps. 

For all these calculations, I am using this website: https://www.cellmapper.net/4G-speed. I am using SISO and band 7, 64 QAM down and 16 up.

When I test the uplink with iperf in UDP mode, I get much closer to the expected 12 Mpbs, and the rate is much more stable. I also tested the uplink and downlink rates on the ZMQ srsRAN 4G setup I have on my laptop, and I got the expected rates with iperf3 in TCP mode. The top picture below shows relevant metrics from the run on the real lab setup, and the one below shows the ZMQ simulation version. 

[image]
[image]

Logs and config files for the lab run are attached below:
[github_iperf_issue.zip](https://github.com/user-attachments/files/17937285/github_iperf_issue.zip)

## Setup Details ##

[UE server] <-> [USRP N310] <--RF--> [USRP N310] <-> [eNB server]

I checked both UE and eNB, and CPU usage is not an issue here.

## Expected Behavior ##
I expect to see ~10 Mbps on the uplink using iperf3 TCP.

## Actual Behaviour ##
I am seeing data rates with large variation well below 10 Mbps. Its worth mentioning that sometimes I do see the correct 10 Mbps iperf TCP on the uplink, but it is inconsistent. When it does work, it is usually after running iperf for a while, or running the TCP iperf right after the UDP iperf.

## Steps to reproduce the problem ##
Configuration files are attached above. Running normal iperf3 on the uplink should reproduce this issue.

## Additional Information ##
I have a theory, which is that there is some issue between TCP and the uplink resource allocation process in the eNB. From my limited understanding of TCP, I believe that it starts with a low rate, then increases until it starts getting lost packets, at which point it slows down. At the same time, the eNB resource allocation is dependent on how much data is in the UE buffer waiting to be sent. So I could image in situation in which TCP tried to send too fast, and somehow the packets get lost and then TCP backs down. While this happens, the eNB might see the increase in buffer and allocate more resources, but when the resources are available to the UE, TCP has already backed down, and so then the eNB reduces allocation. Not a fully fleshed out idea, but it seems like this could happen?

matth3wh, 2024-12-17:
Update: I have been digging into this for a few weeks now and I have a few more pieces of information. When I run IPERF UDP on the uplink at rates that should be fine (5Mbps, 8Mbps ...) I get a fluctuating 2-4% rate of dropped packets. I believe this is the cause of the poor TCP performance. @andrepuschmann if you have a second could you check this out? Thanks!

andrepuschmann, 2024-12-17:
Before trying to optimize TCP I would first make sure everything works as expected with UDP. You say you get dropped packets - but where? At app-level? If you send more packets than the channel capacity iperf will report dropped packets - that's expected.

matth3wh, 2024-12-17:
Thanks for the reply! 

I agree, getting UDP working seems like the first step here. I do not believe that the dropped packets are because of the channel capacity. I believe the channel capacity is ~10Mbps, and when I send UDP at 15Mbps I get a steady 33% loss rate, which makes sense. However, when I send UDP at 5 Mbps, which is well under the channel capacity, I get a 2-4% fluctuating packet loss rate. 

I'm unfortunately not super familiar with networking, all I know is that the IPERF server I am running on the EPC machine is reporting UDP packet loss. And at the same time, the link conditions (SNR, BLER) seem to be great. 

If there were buffers overflowing within srsRAN, would the software spit out errors in the logs? Otherwise I am not sure how to proceed in debugging, if you have any suggestions I would welcome them!

matth3wh, 2024-12-18:
Sure thing! Here are the screenshots. In order, they are:

1) UE metrics trace
2) eNB metrics trace
3) UE iperf client trace using udp at 5M
4) eNB iperf server trace

[image]
[image]
[image]
[image]

Some observations I have:

1) It looks like the UE transmits at a rate _slightly_ higher than the eNB receives at, is this expected? I also observe this behavior running the system in ZMQ mode, so I am assuming it is okay.
2) SNR looks fine. I should note that even though in these screenshots the MCS is 22, I have observed the exact same behavior when the MCS is 23. 
3) I don't see any errors reported by the eNB. 

I have also attached the logs (set to warning). Not much interesting here, except one error in the eNB.
[enb.log](https://github.com/user-attachments/files/18183467/enb.log)
[ue.log](https://github.com/user-attachments/files/18183469/ue.log)

matth3wh, 2024-12-19:
Figured out the issue! It turned out that 2-4% of the traffic sent from the eNB to the EPC was lost in out lab's network switch. Not exactly sure why yet, but connecting the UE computer, eNB computer and EPC computer via a smaller, dedicated switch solved the issue. 

I discovered this by summing the number of GTPU packets leaving the eNB per second and comparing that to the number of GTPU packets entering the EPC. This is where I noticed the loss of packets. 

I calculated the GTPU packet rate using the logs from the eNB and the EPC, for example:

2024-12-18T15:07:08.486902 [GTPU   ] [I] Tx S1-U SDU, UL > 10.33.220.129:0x1, rnti=0x46, eps-BearerID=5, n_bytes=52, IPv4 172.16.0.2 > 172.16.0.1

This resolves my issue and I hope it can be helpful to anyone who comes across it in the future!

closed 2024-12-19 by matth3wh

BENIEJAISON, 2025-03-01:
I am observing some unexpected pattern in data rate in iperf for bandwidth of 100 MHz-150 Mbps (sampling rate - 122.88 Msps) 60 MHz - 135 Mbps (sampling rate 61.44 Msps) but i expect more data rate in 100 MHz what is the reason

i am using USRP N310 and i7 processor

BENIEJAISON, 2025-03-01:
The above problem is for srsRAN_Project and n78 band of frequency 3.5GHz