srsran/srsRAN_4G #1476: Make failure for latest version

MalinkyZubr, 2025-05-08:
Hi,
I am trying to install srsRAN_4G version 23.11 on ubuntu 24.04, and am receiving the following errors

[image]

[image]

I have (as far as I can tell) followed all installation instructions listed in https://docs.srsran.com/projects/4g/en/latest/general/source/1_installation.html. I installed all listed dependencies.

I wasn't able to find this issue already posted, but if it has been solved in the past please redirect me.

Thanks

andrepuschmann, 2025-05-09:
Hey, compiles for me on Ubuntu 24.04. What is your gcc version?
```
$ gcc -v                                                                                                                                                                                                                                                                                                                                    127 ↵
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-linux-gnu/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 13.3.0-6ubuntu2~24.04' --with-bugurl=file:///usr/share/doc/gcc-13/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-13 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/libexec --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-libstdcxx-backtrace --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-13-fG75Ri/gcc-13-13.3.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-13-fG75Ri/gcc-13-13.3.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.3.0 (Ubuntu 13.3.0-6ubuntu2~24.04)
```

s5uishida, 2025-05-11:
Hi @MalinkyZubr , @andrepuschmann 

On Ubuntu 24.04, I downgraded the gcc version from 13 to 11, and was able to build and run the latest version of srsRAN_4G. Also, I got compilation errors with gcc version 12 or 13.

For details, see below.

https://github.com/s5uishida/build_srsran_4g_zmq_disable_rf_plugins

FYI.

andrepuschmann, 2025-05-11:
@s5uishida @MalinkyZubr Apologies - I've just noticed I've tried with our internal branch which has the gcc 13 issue already fixed. We'll push the update very shortly.

closed 2025-06-18 by MalinkyZubr